PSP-Archive/ARK-4 #504: Custom Launcher crashes when highlighting certain *.cso game item in the list

gezzerlol, 2025-02-20:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.69 r163

### Homebrew/GAME (if applicable)

Custom Launcher

### Plugins ( if applicable )

no PLUGINS.TXT

### Steps to reproduce

1. I boot the PSP
2. I launch Custom Launcher
3. I press down to highlight a certain item
4. PSP freezes, after a few second it turns off

At first I thought this is something game image specific but as I deleted the images that crash the launcher, the items before the deleted ones started to crash the launcher, makes no sense to me. EBOOT homebrews and PSX games cause no issue. I'm willing to provide any debug information if present.

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-03-16:
The game is most likely corrupt. Download/copy again.

closed 2025-03-16 by JoseAaronLopezGarcia

gezzerlol, 2025-03-17:
The game surely is corrupt. The reason I've opened this as a issue is that the launcher shouldn't crash when encountered a corrupt image, instead it should show "corrupt file" or ignore the file completely. I think this is incorrect behaviour.

JoseAaronLopezGarcia, 2025-03-18:
I understand this, but it's very difficult to debug this kind of stuff unless it's happening to me and even then my corrupted file can cause a totally different crash than yours, and the crashing code might not be mine but any of the many third party libraries I use. So it's hard to reproduce the exact crash you had.

It would have been much easier if you could have provided some psplink output, but I don't expect common users to even know how to use that tool.