PSP-Archive/ARK-4 #505: WipeOut Pure crashes when Medievil teaser is run (80010016)

michaelvernon, 2025-02-21:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

163

### Homebrew/GAME (if applicable)

UCES-00001

### Plugins ( if applicable )

none

### Steps to reproduce

1. open wipeout pure
2. navigate to teasers menu
3. open medievil teaser
at this point, if "use extra memory" is on AUTO, the game crashes with a "This game could not be started. 80010016" error. if "use extra memory" is on FORCED, the PSP shuts itself off. I have verified this error still occurs regardless of inferno cache setting

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

xeonliu, 2025-02-21:
similar issue after updating.
downgraded to 30th anniversary release and it works

JoseAaronLopezGarcia, 2025-02-21:
> similar issue after updating. downgraded to 30th anniversary release and it works

So it's a regression. Any idea when it might have been introduced?

michaelvernon, 2025-02-21:
These reddit threads discuss similar issues: https://www.reddit.com/r/PSP/comments/1d46opj/select_umd_isos_crashing_on_psp/
https://www.reddit.com/r/PSP/comments/1hvzhlm/sotn_not_launching_on_dracula_x_chronicles/
https://www.reddit.com/r/PSP/comments/1i92qxa/error_80010016/

krazynez, 2025-02-21:
[image]

Working fine on 163 for me

JoseAaronLopezGarcia, 2025-03-16:
Closing as it can't be reproduced.

closed 2025-03-16 by JoseAaronLopezGarcia