PSP-Archive/ARK-4 #513: [Custom Home Mod PSP] ID is empty for ISOs launched from `CL`

Aneemasta, 2025-03-11:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.69 r163 [v1740837232](https://github.com/PSP-Archive/ARK-4/tree/v1740837232)

### Plugins ( if applicable )

psp homebrew, ms0:/SEPLUGINS/cushome/cushome.prx, on
launcher, ms0:/SEPLUGINS/cushome/cushome.prx, off
xmb, ms0:/SEPLUGINS/PSPconsole/PSPconsole.prx, on
xmb, ms0:/SEPLUGINS/xmbih/xmbih.prx, on
ps1, ms0:/SEPLUGINS/cdda_enabler.prx, on
always, ms0:/SEPLUGINS/sssss/sssssbmp.prx, on
always, ms0:/SEPLUGINS/sssss/ssssspng.prx, off
ULJM05472, ms0:/SEPLUGINS/divapatch/divapatch.prx, on
NPJH50465, ms0:/SEPLUGINS/divaxpatch/divapatch.prx, on

### Steps to reproduce

#### Prerequisites:
- [Custom Home Mod PSP](https://github.com/PSP-Archive/ARK-4/wiki/Recommended-Plugins#custom-home-mod-psp) plugin installed for `GAME` or `UMD/ISO`
#### STR:
1. Boot into `XMB`
2. Run any ISO game
3. Press `HOME` key to open `Custom Home Mod PSP`
4. Navigate to `PSPStates` menu item with `D-Pad` and press `X`
5. Press `O`, then navigate to `Exit Game` menu item with `D-Pad` and press `X`
6. Run `Custom Launcher`
7. Repeat steps from 2 to 4
#### What was expected:
- `Custom Home Mod PSP` and `PSPStates` menus display game ID regardless of where the game launched from
#### What was happened:
- `Custom Home Mod PSP` and `PSPStates` menus display zero-length game ID when the game launched from `CL`
#### Screenshots:
Haven't managed how to actually *capture* the `HOME` screen, so here are some photos with those shiny reflections of my hands 🤦‍♂️
1. When launched from `XMB`
[image]
2. When launched from `CL`
[image]
---
P.S.: This issue is irrelevant for homebrew from `/PSP/GAME/`, like Retroarch.

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

On

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

On

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

krazynez, 2025-03-11:
Hmmm interesting perhaps because its launching a game within a game, but oddly enough it resets the device the same way when doing it. The only thing that currently comes to my mind is perhaps apitype but that is just quick thinking. I will have to try to see if the same happens on pro or not. Do you know if there is src for custom HOME? Can always check to see how that its getting the game id's and see if we are doing something on our side wrong. But again I have to verify if PRO has same issues or not. 

I always like your bug reports very clean steps and structure which is very appericiated.

Aneemasta, 2025-03-11:
> I always like your bug reports very clean steps and structure which is very appericiated.

My primary occupation is QA, so it's just natural. But thanks anyway 😊

Aneemasta, 2025-03-11:
> its launching a game within a game

Probably worth mention that `Castlevania: The Dracula X Chronicles` behaves similar when launches integrated PS1 ports of `Rondo of Blood` and `Symphony of the Night`. `PSPStates` menu then displays `ISO` prefix for future savestates.

JoseAaronLopezGarcia, 2025-03-18:
Try latest pre-release.

krazynez, 2025-03-18:
[image]

None pre-release

Aneemasta, 2025-03-20:
Tried to launch `ULKS-46155` / [Castlevania: The Dracula X Chronicles](https://vimm.net/vault/28682) with [v1742383608](https://github.com/PSP-Archive/ARK-4/tree/v1742383608) and this time PSP crashes while attempting to launch the game, both from `XMB` and from `CL`. Rolling back to `r164` for now.

JoseAaronLopezGarcia, 2025-03-20:
Try this one: https://github.com/PSP-Archive/ARK-4/releases/tag/v1742475564

JoseAaronLopezGarcia, 2025-03-20:
On latest (pre)release I have no issues, Custom Home's savestate plugin detecs Game ID just fine.
Closing this.

closed 2025-03-20 by JoseAaronLopezGarcia