PSP-Archive/ARK-4 #514: using the "cut" option in custom launcher file browser on folders deletes them

kay5421, 2025-03-13:
### PSP/Vita Model

PSP-E1000 (Street)

### ARK Version

r163

### Steps to reproduce

1. open up the file browser
2. select a few folders
3. use the "cut" option
4. try to paste the folders
5. pop-up shows up "a directory already exists. what to do? 1. cancel 2. change name 3. overwrite"
6. choosing any option starts the deletion process of the selected folders

### USB Charge

On

### CPU Clock in Game

Overclock

### Power Save

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Use Extra Memory

Forced

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

On

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

Aneemasta, 2025-03-14:
+1: #450

JoseAaronLopezGarcia, 2025-03-16:
Fixed in latest (pre)release.

closed 2025-03-16 by JoseAaronLopezGarcia