PSP-Archive/ARK-4 #515: ARK Loader hanging on Executing ARK Stage 2

kristiannowak, 2025-03-14:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

Rev 163: Unexpected Update

### Homebrew/GAME (if applicable)

ARK Loader

### Plugins ( if applicable )

none

### Steps to reproduce

1. Copy ARK_01234 to SAVEDATA
2. Copy ARK_Loader to GAME
3. Copy ARK_cIPL to GAME
4. Copy ARK_Full_Installer to GAME
5. Run ARK_Loader

### USB Charge

Off

### CPU Clock in Game

Auto

### Power Save

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

krazynez, 2025-03-14:
Could be corrupt card, or perhaps corrupt files. I would redownload and delete ARK_01234 and ARK_Loader and recopy them and safley eject from PC. See if that works.

kristiannowak, 2025-03-15:
redownloading and recopying worked!

closed 2025-03-15 by kristiannowak