PSP-Archive/ARK-4 #517: Option to disable internal storage on PSP Go

kuroxtyan, 2025-03-15:
### New Feature

This option is essential for PSP Go owners, like air to breathe. Due to internal storage corruption (error 802E100A), the system constantly tries to access the internal memory (EF0:/), making it impossible to use Chronoswitch. Please add the ability to use only MS0:/ – this would be an amazing exclusive feature for the ARK firmware. You will save millions of consoles! :)

krazynez, 2025-03-15:
Why are you trying to run Chronoswitch if your already running ARK?

kuroxtyan, 2025-03-15:
This is a simple example of what is happening. But the most important thing is that the system constantly refers to this section.

JoseAaronLopezGarcia, 2025-03-15:
I have been thinking about this for a while and I think I know how to do it.

JoseAaronLopezGarcia, 2025-03-16:
TRY THIS AT YOUR OWN RISK! I HAVE NOT TESTED THIS!
MAKE SURE YOU HAVE NEW CIPL PREVIOUSLY INSTALLED IN CASE SOMETHING GOES WRONG YOU CAN RECOVER THE SYSTEM VIA DC10!!!!!!

Instructions:
- Open Advanced VSH Menu or Recovery Menu
- Change USB Device to "flash0" and (in VSH Menu) make sure USB ReadOnly is disabled.
- Connect PSP to PC, you should see a folder named "kd".
- Copy the two files inside the zip (they should be deadef.prx and pstbtcnf_05g.bin). There are NEW files, there should be absolutely NO message telling you to replace anything.
- Reboot the device and cross your fingers.

[deadef.zip](https://github.com/user-attachments/files/19271780/deadef.zip)

kuroxtyan, 2025-03-18:
This is very risky. What steps should I take to protect myself?

krazynez, 2025-03-18:
He said what, DC10 [DCARK](https://github.com/PSP-Archive/ARK-4/wiki/DC%E2%80%90ARK)

JoseAaronLopezGarcia, 2025-03-18:
> This is very risky. What steps should I take to protect myself?

I already told you, install New cIPL (which you should have already) so you can recover via DC10 by holding L trigger when powering on the console. DC10 can be installed with a PC so you don't even need a functional PSP.

Look mate, I have no PSP Go with broken internal memory so I can't test this for you, so it's up to you to verify that my idea works or not. The risk is nonexistent (if you have New cIPL installed, which you should) and the rewards are very important.

Otherwise I'll just close this thread if it's not going anywhere.

kuroxtyan, 2025-03-18:
Okay, I'll try and write the result.

kuroxtyan, 2025-03-18:
Ok I tried: After writing files to flash0, reboot, black screen, green LED lights for 1 minute, then the console turns off. The console can be turned on by loading ARC from a flash card via DC

JoseAaronLopezGarcia, 2025-03-18:
Alright I'm gonna review the code and try things. Stay tuned.

kuroxtyan, 2025-03-18:
> Хорошо, я проверю код и попробую разные штуки. Оставайтесь с нами.

okay

JoseAaronLopezGarcia, 2025-03-18:
yeah I've found a few issues in pstbtcnf, give me a minute.

JoseAaronLopezGarcia, 2025-03-18:
Try this one:

[deaf.zip](https://github.com/user-attachments/files/19319264/deaf.zip)

You should be able to use DC itself to replace the files without having to reinstall the whole firmware.
Just select the option to "Boot ARK-4 from Memory Stick", from there you can do the same (Advanced VSH Menu -> change USB device to flash0, etc).

kuroxtyan, 2025-03-18:
> Try this one please:
> 
> [deaf.zip](https://github.com/user-attachments/files/19319264/deaf.zip)
> 
> You should be able to use DC itself to replace the files without having to reinstall the whole firmware. Just select the option to "Boot ARK-4 from Memory Stick", from there you can do the same (Advanced VSH Menu -> change USB device to flash0, etc).

Okay, now the console doesn't turn off, but it also doesn't turn on, the screen backlight just lights up, and it often accesses the external memory card

JoseAaronLopezGarcia, 2025-03-18:
Interesting. It could be a bit more complex then.
Lets try this and see what happens:

[deadef.zip](https://github.com/user-attachments/files/19319733/deadef.zip)

kuroxtyan, 2025-03-18:
> Interesting. It could be a bit more complex then. Lets try this and see what happens:
> 
> [deadef.zip](https://github.com/user-attachments/files/19319733/deadef.zip)

file pstbtcnf_05.bin, should I also place it in the folder or delete it? The system without the file pstbtcnf_05.bin - starts, but the internal memory is visible in XMB

JoseAaronLopezGarcia, 2025-03-18:
keep that file, only replace what I send you, but you need the two files.

kuroxtyan, 2025-03-18:
> keep that file, only replace what I send you, but you need the two files.

the result is 1:1 as in the previous case, the screen backlight just lights up

kuroxtyan, 2025-03-18:
Maybe I'm doing something wrong?

JoseAaronLopezGarcia, 2025-03-18:
I'm gonna do some tests myself on my own PSP Go, though my Go has a healthy internal memory so it might not be the best test scenario, but I will at least try to get something to actually boot and see if I can have internal memory disabled on my Go.

kuroxtyan, 2025-03-18:
> I'm gonna do some tests myself on my own PSP Go, though my Go has a healthy internal memory so it might not be the best test scenario, but I will at least try to get something to actually boot and see if I can have internal memory disabled on my Go.

this is just cool, if you need more tests, let me know

JoseAaronLopezGarcia, 2025-03-18:
I think I got it, please stand by.

JoseAaronLopezGarcia, 2025-03-18:
This "works" for me, it prevents ef0/eh0 drivers from running and shows the XMB as if it weren't a PSP Go.
Then again I have a heathy PSP Go so I don't know if it'll be enough for you.

[deadef.zip](https://github.com/user-attachments/files/19321890/deadef.zip)

kuroxtyan, 2025-03-18:
> This "works" for me, it prevents ef0/eh0 drivers from running and shows the XMB as if it weren't a PSP Go. Then again I have a heathy PSP Go so I don't know if it'll be enough for you.
> 
> [deadef.zip](https://github.com/user-attachments/files/19321890/deadef.zip)

Okay, that's cool, the console started!!! But it can't display the XMB menu items

JoseAaronLopezGarcia, 2025-03-18:
Try disabling all plugins and make sure you have ARK_01234 installed in external memory card.

JoseAaronLopezGarcia, 2025-03-18:
If things get too ugly you might have to forget about the XMB and use Custom Launcher instead, there's a CFW option to autoboot launcher.

kuroxtyan, 2025-03-18:
> 

Yes, the custom launcher works and displays games! Previously, when ef0 was active, the custom launcher couldn't launch at all!
Also, please note: if you try to pause the game using the PS button, the console shuts down — therefore, the ability to pause the game must be disabled.

JoseAaronLopezGarcia, 2025-03-18:
There's already an option to Disable Pause Feature in the CFW.

JoseAaronLopezGarcia, 2025-03-18:
Btw from now on you can use ARK_Full_Installer to install/uninstall the DeadEf mod.

closed 2025-03-18 by JoseAaronLopezGarcia

kuroxtyan, 2025-03-18:
> 
should plugin support on ef0 be disabled too?