PSP-Archive/ARK-4 #518: native support for folders

pcalkovsky, 2025-03-17:
### New Feature

I'd love to see some better form of folders/sorting than categories lite, since this plugin is not that great to use especially on psp go (the plugin can crash psp go when trying to create new save file).
Would it be possible to have folders with shortcuts on memory stick/internal storage (perhaps .lnk files)? That way we would have xmb sorted, while the isos would still be in ISO directory on memory stick/internal storage.

JoseAaronLopezGarcia, 2025-03-17:
We already have native folder support in the Custom Launcher. Either use the "Scan Categories Enties" option, or use the File Browser as your main app launcher.

pcalkovsky, 2025-03-17:
That is great, thank you, but is something like this possible in XMB as well ?

JoseAaronLopezGarcia, 2025-03-17:
It is possible I guess, but that's something for a plugin to do, not for the CFW.
At some point adding more stuff to the XMB adds more to the boot times, and some people might prefer to have the option to install more things (via plugins) or keep a minimal system.

closed 2025-03-17 by JoseAaronLopezGarcia