PSP-Archive/ARK-4 #519: Misc ad-hoc related things

Double-0-seven7, 2025-03-18:
### New Feature

Putting this as a feature request because I am pretty sure most if not all of the following happen on Vita pspemu/Adrenaline as well.
Here are the following games/features I found that got issues working on a Vita:

- [ ] Games that use GameMode API - partial list of games [here](https://report.ppsspp.org/logs/kind/1093).
- [ ] Burnout Dominator - Game becomes too fast for everyone if a Vita player joins , some report says it also applies to the PSP 1000 model as well (video demonstration [here](https://www.youtube.com/watch?v=Fbeaqwpj-lU)).
- [ ] Lego Star Wars II - Can't connect to other Vitas/PSPs
- [ ] Street Riders/Asphalt: Urban GT 2 - Can't see other consoles in lobby (same developer made both games so its most likely the same issue).
- [ ] Games that use GameSharing - Not 100% sure about this one but I think it only works on regular pspemu for now.

Related issues : https://github.com/PSP-Archive/ARK-4/issues/406 and https://github.com/PSP-Archive/ARK-4/issues/32.
This only applies to Vita consoles , PSP consoles should be fine except for possibly Burnout Dominator (PSP 1000) and Street Riders/Asphalt Urban GT 2 (I remember getting the lobby issue on PSP Go as well).

Double-0-seven7, 2025-03-18:
Tried Gamesharing right now and it seems to work fine so I replaced that issue with another one that is related to the homebrew CSPSP.

Double-0-seven7, 2025-04-02:
Lego Indiana Jones 2 got the same issue as Lego Star Wars 2.
Curiously on ARK4 when you host a game the whole game freezes but on adrenaline you get a wireless error instead.
I wonder how it behaves on normal pspemu now.

JoseAaronLopezGarcia, 2025-04-02:
Some of these could perhaps be fixed with UMD delay option?

> Lego Indiana Jones 2 got the same issue as Lego Star Wars 2. Curiously on ARK4 when you host a game the whole game freezes but on adrenaline you get a wireless error instead. I wonder how it behaves on normal pspemu now.

ARK standalone should behave the same (or similar to) original pspemu.

JoseAaronLopezGarcia, 2025-04-02:
> BTW i had to use a specific ARK4 version because the latest development/stable one is not working right now.

Can you elaborate on what's broken?

JoseAaronLopezGarcia, 2025-04-04:
We have changed the name of the config. Instead of "delayumd_" we now have "umdspeed_" and "umdseek_", where "umdspeed_" behaves the same as the previous "delayumd_" config.

Double-0-seven7, 2025-04-04:
Just a note, the Lego Star Wars 2 issue could be worked around if you go to the adhoc menu and either turn the wifi switch on and off or go into standby/sleep mode and then back into gameplay.

Double-0-seven7, 2025-04-23:
Compared Burnout Dominator between the PSN/EBOOT version and on ARK Standalone.
PSN - Game is mostly in normal speed with some minor speedups here and there.
ARK Standalone - way over full speed and its way too fast to play at.
So yeah there is some sort of bug here on ARK that is not on the official PSN version here.

JoseAaronLopezGarcia, 2025-04-23:
Did you try enabling UMD speed?

JoseAaronLopezGarcia, 2025-04-25:
> > Did you try enabling UMD speed?
> > El mié, 23 abr 2025, 20:30, Double-0-seven7 _**@**_.***>
> > escribió:
> > […](#)
> 
> No point as I run the game as eboot in both places (umd delay/speed wont work).

So you're saying that the same PSN EBOOT when ran alone will be a normal speed but with ARK it's "too fast"?
Well I can't really understand why you'd label that a "bug", it's obviously the result of OverClock and Memory Stick Speedup, most likely OverClock.

Double-0-seven7, 2025-04-25:
Overclock was on both "Auto" and "Powersave" and it was still too fast.
I didn't try the memory stick speedup thing though.

JoseAaronLopezGarcia, 2025-04-25:
> I didn't try the memory stick speedup thing though (does it work on

Just convert the eboot to an ISO amd use the delay feature. Yes Ms speedup
works on eboots but I very much doubt it's the problem.
It actually makes no sense for me to think that ARK would run faster than
stock pspemu since they are both the same.

Double-0-seven7, 2025-04-25:
Tried memory stick speedup and it was still the same hmmm.
I saw that this issue also happens on PPSSPP and there is a developer that mentioned a synchronization issue here : https://github.com/hrydgard/ppsspp/issues/14661#issuecomment-894088770
Its for NFS Shift but its by the same developer and exhibits the same issue as Dominator
V-sync is also mentioned here : https://github.com/hrydgard/ppsspp/issues/13452#issuecomment-695832588
Could it be a vsync difference between stock pspemu/PSN and ARK standalone perhaps?

JoseAaronLopezGarcia, 2025-04-25:
ARK Standalone works on top of stock pspemu, there's no difference there. You can literally see the ChovySign (#ChovyProject) boot screen.
Only a single plugin called ark_right_analog patches right analog input into ARK, but that's totally unrelated, in if anything I would expect people to scream "input lag" rather than "it's running faster".

Again, try the ISO and use the UMD Delay settings (i.e. umdspeed).

Double-0-seven7, 2025-04-25:
> ARK Standalone works on top of stock pspemu, there's no difference there. You can literally see the ChovySign (#ChovyProject) boot screen.

Its still not a 1:1 experience though.
Try running Twisted Metal Head On you got from PSN and on ARK Standalone.
On PSN you will be missing 2 options that are related to infrastructure (because presumably they broke it like mentioned in another issue) while ARK will retain them.
Both use the same EBOOT.

[image]

[image]

JoseAaronLopezGarcia, 2025-04-25:
This is most likely due to the fact that ARK Loader (the ChovySign-generated EBOOT.PBP) uses some fake Game ID.
Stock pspemu uses the ID of the game to configure itself. ARK doesn't use any ID from any game, so pspemu probably uses stock config (hence why ARK has to fix Star Wars Force Unleashed while the PSN version doesn't need a fix).

Double-0-seven7, 2025-08-03:
A bit late but the UMD Delay option doesn't help with Burnout Dominator speedups.
So yeah you have to run the game from PSN as is without either ARK or Adrenaline.

JoseAaronLopezGarcia, 2025-08-03:
> A bit late but the UMD Delay option doesn't help with Burnout Dominator speedups. So yeah you have to run the game from PSN as is without either ARK or Adrenaline.

What if you run the PSN version with ARK or Adrenaline?

GrayJack, 2025-09-24:
It is an synchronization issue. ePSP implements `sceWlanGetSwitchState` with cache (it caches the value and the saves the time of the cache), but only locks the mutex arround the kermit request call. I went to the route of reimplementing `sceWlanGetSwitchState` without the cache and didn't cause a major slowdown, but reimplementing with cache and syncing correctly is also an option.

Double-0-seven7, 2025-09-27:
> > A bit late but the UMD Delay option doesn't help with Burnout Dominator speedups. So yeah you have to run the game from PSN as is without either ARK or Adrenaline.
> 
> What if you run the PSN version with ARK or Adrenaline?

It still happens regardless , I think we talked about it being a patch that is not loaded on either ARK or Adrenaline but I am not sure.
BTW the Sega Rally Revo issue is fixed by using this plugin : https://github.com/Kethen/pspemu_inet_multithread 
Still would be nice if there was a non plugin way to make it work.