PSP-Archive/ARK-4 #522: Real UMD delay as a toggle

afrowebster, 2025-03-21:
### New Feature

This is an amazing feature, first of all hats off to you.
However I think it could be even more useful for some multiplayer games as an option that could be turned on or off, as one system is always loading faster than the other making them disconnect.
This is absolutely a problem in the GTA Liberty City Stories game.
Potentially this could fix this ancient problem.

Double-0-seven7, 2025-03-22:
To be frank I only saw a multiplayer issue in that 1 game you described (and maybe VCS too).
I am playing a lot of multiplayer games and never got issues that depend on UMD timings.

afrowebster, 2025-03-22:
Including the IDs of these games might also work but in that case it would be just slower for 99% of the players without any benefit.
That's why I thought about a toggle.

JoseAaronLopezGarcia, 2025-03-23:
I don't like the idea of an easy toggle, I already see users complaining that "ARK runs games slower than PRO/ME" simply because they have this turned on. Same thing happened when ARK used to not enable cpu overclock by default, stupid people kept complaining that "performance was slower than PRO/ME" where overclock was default and they were too stupid to check settings.
What I've done is I have added the config to the CFW internally but I'm not gonna have it show as in any GUI.
You will have to manually edit SETTINGS.TXT and add your own rules for whichever game you want to run with UMD Delay.

# Enable UMD Delay in LCS and VCS
ULUS10041 ULUS10160 ULES00151 ULES00502, delayumd_2, on

You can use delayumd_1 for 1MB/s (enough to fix Aces of War but not enough for DJMax) or delayumd_2 for 512 KB/s (closest to actual UMD speed and enough to fix DJMAX). Higher numbers will yield slower reading speeds.

closed 2025-03-23 by JoseAaronLopezGarcia

afrowebster, 2025-03-23:
Editing the SETTINGS.TXT is perfectly fine for me.
Is it on the memory card or in one of the flashes?

JoseAaronLopezGarcia, 2025-03-23:
> Editing the SETTINGS.TXT is perfectly fine for me. Is it on the memory card or in one of the flashes?

By default it's located in ARK_01234 savedata folder. If you don't have that folder then SEPLUGINS will work too.
We don't recommend flash settings like other CFW do (we allow it, to some degree, but don't encourage it).

JoseAaronLopezGarcia, 2025-03-23:
Btw I haven't really tested it, but the code was simple enough that it should work. If there's any issues just let me know.

afrowebster, 2025-03-23:
Awesome thank you.
Right now I'm charging 2 of my PSPs to test this :D
I'll come back with the results.

afrowebster, 2025-03-23:
Unfortunately no luck.

The host machine always loads way faster than the client.
I've tested it with a 1000 and a GO, no matter which hosted it was always faster.

I've removed the highmem setting and went with this:
# Enable UMD Delay in LCS and VCS
ULUS10041 ULUS10160 ULES00151 ULES00502, delayumd_2, on

JoseAaronLopezGarcia, 2025-03-23:
Do you percieve an actual change in the load times of LCS? It could be that I have made a mistake with the setting.

afrowebster, 2025-03-23:
It is really hard to tell as it loads really quickly, around 3 secs with or without this setting. (no significant difference)
Also should it work with R167?
Maybe I just need to download a "nightly".

JoseAaronLopezGarcia, 2025-03-23:
There should be a significant difference between loading GTA LCS on an ISO vs a UMD. This sounds like the setting isn't properly implemented perhaps.

afrowebster, 2025-03-23:
Maybe I'm doing something wrong.
I have edited the SETTINGS.TXT file in the SAVEDATA folder.
This is how it looks like:
[https://pastebin.com/fBJkFvaR](url)

JoseAaronLopezGarcia, 2025-03-23:
There was a bug in the code, give it some minutes to generate a new pre-release and update to it.

afrowebster, 2025-03-23:
Will do ;)

afrowebster, 2025-03-23:
Sir, we are onto something.
Now the load times were significantly slower, although hosting the game on the GO resulted in the same problem as before, when hosted on my OG 1000 it actually worked.
[PROOF](https://imgur.com/a/0sCffgj)
I cannot express how awesome you and your project is. Please accept my greatest gratitude, you just solved a very niche issue tainting the multiplayer scene from the beginning.

JoseAaronLopezGarcia, 2025-03-23:
Perhaps this could also solve the Ad-Hoc issues with PS Vita.

afrowebster, 2025-03-23:
I think I have maxxed this out.
4 players:
PSP-1000
PSP-2000
PSP GO
PS Vita TV
Working perfectly, even the VITA can host.
[I cannot believe it is working](https://imgur.com/a/6RoVE15)

Thank you, thank you, thank you!!!

JoseAaronLopezGarcia, 2025-03-23:
Holy shit that's actually pretty awesome!

JoseAaronLopezGarcia, 2025-03-24:
> I think I have maxxed this out. 4 players: PSP-1000 PSP-2000 PSP GO PS Vita TV Working perfectly, even the VITA can host. [I cannot believe it is working](https://imgur.com/a/6RoVE15)
> 
> Thank you, thank you, thank you!!!

For solving Ad-Hoc sync issues perhaps delayumd_1 is enough, so you don't have to resort to those horrible UMD speeds. 1MB/s is closer to DVD speed than UMD speed and I believe as long as they are all using the same loading times it'll be fine. Perhaps you might want to test this.

afrowebster, 2025-03-25:
I'll certainly test it around Wednesday night. 
I'm away from home atm.

Double-0-seven7, 2025-03-25:
> Perhaps this could also solve the Ad-Hoc issues with PS Vita.

Not ad-hoc but I tried the infra again with Sega Rally Revo and it didn't work.
Tried using this line : 
`ULUS10311, delayumd_2, on`
But still got a disconnection after race start.
BTW I suppose this doesn't work in PBP format? I tried it in both PBP and ISO format but didn't notice loading times changing much.
I also don't think its umd timings related anyway because PPSSPP can play the game online without using that option enabled in it's System settings.

afrowebster, 2025-03-27:
Sorry for the late reply, so I wanted to try your theory about umd_delay_1 should be enough however, the umd speed emulation is broken in hte latest release: [v1743091428](https://github.com/PSP-Archive/ARK-4/tree/v1743091428)

JoseAaronLopezGarcia, 2025-04-04:
I decided to replace "delayumd" with two other settings, "umdspeed" and "umdseek". In this context, "umdspeed" behaves the same as "delayumd" used to behave. They are both used the same way ("umdspeed_1", "umdseek_1", etc).