PSP-Archive/ARK-4 #524: When CXMB for VSH is enabled, some games System UI such as when saving show no or weird letters.

Klejdi90, 2025-03-21:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

ARK 4.20.69 r167

### Homebrew/GAME (if applicable)

UCES01250, ULES01144

### Plugins ( if applicable )

vsh, ms0:/seplugins/cxmb/cxmb.prx, on

### Steps to reproduce

1. Enable cxmb, reset vsh.
2. Launch Motorstorm Arctic Edge or Midnight Club LA Remix
3. Attempt to load a save.

[image]

[image]

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-03-22:
CMXB is not supposed to be running on games, only VSH. This is most definitely a user error.

closed 2025-03-22 by JoseAaronLopezGarcia

Klejdi90, 2025-03-22:
I have CXMB set as VSH so why would it run when launching a game?

Klejdi90, 2025-03-22:
Nevermind, sorted was an issue with a specific theme and a game from PKGi
for PSP.

Below is a copy paste from reddit where I asked others for help. Maybe
useful in future developments or if others users show similar problems. If
not sorry for any trouble.

Appears this one game does load CTF theme items even though CXMB is set to
only VSH. So I tried a new memory card and it didn't fix it. However, it
turns out it was the ctf theme I had applied. Here is an image with a
different ctf theme set. https://ibb.co/1GPZYGtB

As you can see, it displays the text ingame, okay. However, note that it
somehow loads the themes font type and different colour for the X and O
icons while in the game? It even plays the themes custom sound when the
prompt says load complete.

My guess is this specific game from PKGi for PSP (Motorstorm Arctic Edge)
maybe a PSN digital game and it appears in PSP>GAME rather than the ISO
folder, loads the XMB system UI and not some Game version of the system UI,
which is why CXMB and by effect the font applies and X/O button colours
even though it is set as a VSH plugin.