PSP-Archive/ARK-4 #527: Game fails to start after update (NPJH50521)

gilgilgan, 2025-03-23:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

Rev 167: UMDelay

### Homebrew/GAME (if applicable)

NPJH50521

### Plugins ( if applicable )

NPJH50521 (super robot wars OE) won't launch after the latest update.

It seemed to launch the first time, but soon it started freezing on the loading screen, and after that, when I restarted the game, I only got a black screen.

I tried rolling back to Rev 163: Unexpected Update, a version that the game previously worked on, and the game ran fine.

### Steps to reproduce

Install arc-4 version 167

Run NPJH50521

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-03-23:
I need much more information if I were to find and fix a potential new bug.

- What type of installation is this? Standalone or Adrenaline-ARK?
- Is the game downloaded from PSN  or by other means?
- Are you running the game from Custom Launcher or XMB?

JoseAaronLopezGarcia, 2025-03-23:
For the record, I just tried a game downloaded from PSN and have no issues launching it on either Standalone via Custom Launcher or Adrenaline-ARK via XMB.

JoseAaronLopezGarcia, 2025-03-23:
Try latest pre-release, it could potentially fix your issue.

gilgilgan, 2025-03-24:
Sorry for not writing down the details
-Standalone
-I got it from another site.
-Custom Launcher

It's fixed in the latest pre-release! Thanks.