PSP-Archive/ARK-4 #528: PSP crashes when trying to load the memory stick game list (iSO folder)

srobycap, 2025-03-23:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

ARK 4.20.69_r167

### Homebrew/GAME (if applicable)

ISO games in ISO Folder. 

### Steps to reproduce

Same bug as #525 

It seems to work if I format again the memory stick. but when I add a game, it freezes. The filesystem seems to be corrupted for the PSP (not the OS, but when a fsck is executed, it fixes few minor errors) 

The #525 bug for reference  (I have only iso files) :  
Store all ISO dumps in the ISO folder as standard, ensure there is no pre-existing ISOCACHE.BIN file in /PSP/SYSTEM, then try and access the Memory Stick item under the Game tab in the XMB. It'll freeze and the system will eventually power off. Custom Launcher is unaffected and doesn't crash this way. I do have manuals and DLC for each game's respective GAMEID located under PSP/GAME/GAMEID, in case that info is relevant. I don't believe this affects PBP/PSN format games however, it seems to be an ISO specific issue.
(I mentioned the ISOCACHE stuff since I could sometimes manage to get the menu to load if that file already existed, but it led to other issues such as the wrong ISO booting for a given selected title in the XMB).

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

dprimenko, 2025-03-23:
Same in PSP-2004 with ARK 4.20.69_r167

JoseAaronLopezGarcia, 2025-03-23:
This just sounds like a broken/corrupted memory stick. In any case, this bug report lacks a lot of essential information.
- How big is the memory stick? 128GB or more will cause this type of issues.
- Do tou have the same issues when launching games from Custom Launcher?
- Does it work if you use an earlier version of ARK?

All in all this really doesn't seem like an ARK issue, otherwise you wouldn't be the only one. I have tons of ISO games on many different models and all work just fine.

I will close this issue unless you can provide with additional information to confirm that it is indeed ARK that's causing issues, then I will reopen it. But ive had enough experience with broken/corruptef memory cards to know that you have thar issue.

JoseAaronLopezGarcia, 2025-03-23:
> > This just sounds like a broken/corrupted memory stick. In any case, this bug report lacks a lot of essential information.
> > 
> > * How big is the memory stick? 128GB or more will cause this type of issues.
> > * Do tou have the same issues when launching games from Custom Launcher?
> > * Does it work if you use an earlier version of ARK?
> > 
> > All in all this really doesn't seem like an ARK issue, otherwise you wouldn't be the only one. I have tons of ISO games on many different models and all work just fine.
> > I will close this issue unless you can provide with additional information to confirm that it is indeed ARK that's causing issues, then I will reopen it. But ive had enough experience with broken/corruptef memory cards to know that you have thar issue.
> 
> Just for reference with my issue [#525](https://github.com/PSP-Archive/ARK-4/issues/525), which this one was a dupe of, my MS is only 32GB and I’m relatively sure isn’t corrupt, CL is unaffected and still works fine as stated in [#525](https://github.com/PSP-Archive/ARK-4/issues/525), and earlier versions of ARK definitely did not have these issues (fairly confident in r163 and earlier being fine from my testing, it’s related to the ISO update/manual changes or something from around then).

I understand and I was expecting new bugs when new features are introduced (like we did with r165, r166 and r167).
But you have to understand that without exact proper instructions to reproduce the issue it's going to be hard, if not impossible, to fix. I have a lot of ISOs and they are all working fine, some have PBOOT.PBP and manuals.

I will need more help in terms of testing. The best way forward is to go to the releases page and download pre-releases.
Every time we change something it geenrates an automatic build (pre-release) and this is very useful for figuring out exactly which commit broke.

I will try to reproduce the issue myself but I have failed so far.

JoseAaronLopezGarcia, 2025-03-23:
I'm going to reopen this thread and close this one: https://github.com/PSP-Archive/ARK-4/issues/528#issuecomment-2746217045

JoseAaronLopezGarcia, 2025-03-23:
> [@JoseAaronLopezGarcia](https://github.com/JoseAaronLopezGarcia) I pinpointed the regression exactly as you said, down to this commit: [70249d5](https://github.com/PSP-Archive/ARK-4/commit/70249d534a5fb2d90b06a5679e6d531346189c0c)
> 
> (Builds I tested as as follows, making sure to delete PSP/SYSTEM/ISOCACHE.BIN between every single run for a fair clean test: //https://github.com/PSP-Archive/ARK-4/releases/tag/v1741971469 works fine //https://github.com/PSP-Archive/ARK-4/releases/tag/v1742073776 works fine
> 
> https://github.com/PSP-Archive/ARK-4/releases/tag/v1742082125 works fine ! https://github.com/PSP-Archive/ARK-4/releases/tag/v1742131032 does NOT work !
> 
> //https://github.com/PSP-Archive/ARK-4/releases/tag/v1742150261 does NOT work //https://github.com/PSP-Archive/ARK-4/releases/tag/v1742161620 does NOT work)

Awesome, it seems to be related to either PBOOT or Manual, but perhaps only a very specific setup? Can you try to move ISOs away until you're left with only the one that breaks?

Currently I have an ISO that has both PBOOT and Manual and it works just fine, perhaps it's ISO + Manual only that breaks it? Or maybe ISO + PBOOT only?

JoseAaronLopezGarcia, 2025-03-23:
Yeah I'm completely incapable of reproducing this issue. It doesn't matter if I have 1 ISO or 50 ISOs, the XMB loads them all just fine. Tried Little Big Planet ISO with PBOOT.PBP but no manual, with manual but no PBOOT.PBP, etc. Can't find a single crash.

> Nailed it down [@JoseAaronLopezGarcia](https://github.com/JoseAaronLopezGarcia), can you try testing Ratchet and Clank: Size Matters' EU version (UCES00420) if you have it? That specific game only consistently causes the XMB to crash with a clean setup (no ISOCACHE.BIN present) with or without a corresponding copy of DOCUMENT.DAT for that game present at all.

That is one of the games I have. I wonder if it has something to do with that ISOCACHE as you mentioned, it's the only thing I haven't done (delete it).

JoseAaronLopezGarcia, 2025-03-23:
> Nailed it down [@JoseAaronLopezGarcia](https://github.com/JoseAaronLopezGarcia), can you try testing an ISO copy of Ratchet and Clank: Size Matters' EU version (UCES00420) if you have it? That specific game and only it consistently causes the XMB to crash for me with a clean setup (no ISOCACHE.BIN present) with or without a corresponding copy of DOCUMENT.DAT for that game present at all.

No luck. I got rid of ISOCACHE.BIN and left Ratchet and Clank Size Matters (EU version) in the ISO folder (along with Little Big Planet and some other ISOs for good measure) and things still work fine.

Could it be just that one ISO that got corrupted for you? But if so then how come it works fine on older builds?

This is getting weirder and weirder by the minute.

JoseAaronLopezGarcia, 2025-03-23:
> > > Nailed it down [@JoseAaronLopezGarcia](https://github.com/JoseAaronLopezGarcia), can you try testing an ISO copy of Ratchet and Clank: Size Matters' EU version (UCES00420) if you have it? That specific game and only it consistently causes the XMB to crash for me with a clean setup (no ISOCACHE.BIN present) with or without a corresponding copy of DOCUMENT.DAT for that game present at all.
> > 
> > 
> > No luck. I got rid of ISOCACHE.BIN and left Ratchet and Clank Size Matters (EU version) in the ISO folder (along with Little Big Planet and some other ISOs for good measure) and things still work fine.
> > Could it be just that one ISO that got corrupted for you? But if so then how come it works fine on older builds?
> > This is getting weirder and weirder by the minute.
> 
> That was something I also suspected, but I went out of my way to MD5 hash my ISO and compare it to Redump's database, and it matches, so it can't possibly be ISO corruption either. I'm going to play around with a few things in a fork and see what part of that commit specifically caused the issue, the PBOOT path fixing or the whole replacement of vpbp_fixmanualpath with the more general vpbp_fixisopath.

Yeah a corrupted ISO is not going to magically work by downgrading ARK version so we can forget about that.

One thing we haven't talked about is plugins. Which plugins do you have installed? It might be possible they break something.

JoseAaronLopezGarcia, 2025-03-23:
> Right, I've pinned down the exact responsible change: it's the PBOOT/update check (is_iso_update), removing that from xmbiso.c magically fixes the crashing somehow (even though Ratchet hasn't ever had any released updates nor do I have a PBOOT file present in PSP/GAME/UCES00420 if that folder even is present during tests).

Awesome! Reading through the code for is_iso_file I've found a potential NULL pointer dereference in line 193.

<code>
    p = strrchr(path, '@') + 9;

    if (0 != strcmp(p, file))
        return 0;
</code>

The value of p is not checked against NULL. I've pushed a fix for this (should generate a pre-release soon). Let me know if that fixes it.

JoseAaronLopezGarcia, 2025-03-23:
Here's the pre-release: https://github.com/PSP-Archive/ARK-4/releases/tag/v1742741663

Let me know if it fixes it.

JoseAaronLopezGarcia, 2025-03-23:
@schm1dtmac did you get to test the pre-release? Does it fix the crash?

JoseAaronLopezGarcia, 2025-03-23:
> > [@schm1dtmac](https://github.com/schm1dtmac) did you get to test the pre-release? Does it fix the crash?
> 
> Sorry for taking a bit longer [@JoseAaronLopezGarcia](https://github.com/JoseAaronLopezGarcia), I had to mess with a few things before testing the new pre-release since the bug stopped occurring for me on the builds from before for some reason. The new release doesn't fix the bug unfortunately, but in the process of getting it to happen again I'm now 100% certain of the cause: macOS. To elaborate, macOS sometimes creates nasty metadata files when copying to non-Apple file systems, so for UCES00420.iso it'll create a metadata file called ._UCES00420.iso when copied to the FAT32 MS filesystem, and that file simply existing causes the PSP freezing issues from before. Explains why only Ratchet had that issue for me since it probably got unlucky and was the only one with that metadata file present, and why you probably didn't have the issue at all (assuming you use Windows like most others).

That is interesting. But why would it happen to OP then? @srobycap do you use MacOS too?
And why would downgrading ARK make it work?
Does this same crash happen on other CFWs like PRO or ME?

Seems like there's a bit more to this than meets the eye.

In any case, perhaps you can attach that strange Mac OS file so I can test it myself and see where and why it causes a crash.

krazynez, 2025-03-23:
> > > > [@schm1dtmac](https://github.com/schm1dtmac) did you get to test the pre-release? Does it fix the crash?
> > > 
> > > 
> > > Sorry for taking a bit longer [@JoseAaronLopezGarcia](https://github.com/JoseAaronLopezGarcia), I had to mess with a few things before testing the new pre-release since the bug stopped occurring for me on the builds from before for some reason. The new release doesn't fix the bug unfortunately, but in the process of getting it to happen again I'm now 100% certain of the cause: macOS. To elaborate, macOS sometimes creates nasty metadata files when copying to non-Apple file systems, so for UCES00420.iso it'll create a metadata file called ._UCES00420.iso when copied to the FAT32 MS filesystem, and that file simply existing causes the PSP freezing issues from before. Explains why only Ratchet had that issue for me since it probably got unlucky and was the only one with that metadata file present, and why you probably didn't have the issue at all (assuming you use Windows like most others).
> > 
> > 
> > That is interesting. But why would it happen to OP then? [@srobycap](https://github.com/srobycap) do you use MacOS too? And why would downgrading ARK make it work? Does this same crash happen on other CFWs like PRO or ME?
> > Seems like there's a bit more to this than meets the eye.
> > In any case, perhaps you can attach that strange Mac OS file so I can test it myself and see where and why it causes a crash.
> 
> OP could also be a macOS user, I also found some redditors who had this same bug who clearly were using macOS: https://www.reddit.com/r/PSP/comments/1jgykj7/game_wont_load_and_crashes_every_time/ (they ended up switching to PRO to avoid the bug) Maybe the file path changes in [70249d5](https://github.com/PSP-Archive/ARK-4/commit/70249d534a5fb2d90b06a5679e6d531346189c0c) trip up when dealing with those weird garbage macOS files? IDK.
> 
> [sidecarfile.txt](https://github.com/user-attachments/files/19409839/sidecarfile.txt) Copy that into your /ISO/ folder on your memory stick, named exactly as ._UCES00420.iso (or some other gameID if testing with another ISO). If File Explorer or whatever won't let you name it as such use terminal or something to force it through (they're invisible on macOS so I had to copy it to something named sidecarfile.txt just to get it to show and upload to GH).

Yes MacOS makes weird hidden folders/files that has caused a lot of issues for many things. I do wonder if there is a way to not parse those, however they are also taking up space on the disk so kinda getting double hit with those.

JoseAaronLopezGarcia, 2025-03-23:
Wait a moment.... If the file ends with .ISO it will try to load it as such, but it is NOT an ISO file.
So the question is, which part of the CFW is crashing due to a file that's obviously not an ISO?
I'll do some digging.

JoseAaronLopezGarcia, 2025-03-23:
Ok so I figured the problem is with the function isoFree, as it tries to close the ISO (cause it's invalid) it is also trying to free pointers that weren't really allocated to begin with. A new pre-release is being generated with a proper fix, hopefully for real this time.

JoseAaronLopezGarcia, 2025-03-23:
Yeah it still crashes, probably needs some more checks here and there. Will let you know when I'm done.

JoseAaronLopezGarcia, 2025-03-23:
Ok I have fixed it. It will be possible that you need to delete ISOCACHE.BIN, in any case now I can see the invalid file as "Corrupted" in the XMB.

JoseAaronLopezGarcia, 2025-03-23:
> > Ok I have fixed it. It will be possible that you need to delete ISOCACHE.BIN, in any case now I can see the invalid file as "Corrupted" in the XMB.
> 
> Yup, latest pre-release solves the issue for me too.

Did you have to delete ISOCACHE.BIN? I did that but I'm not sure if it was really needed.

closed 2025-03-23 by JoseAaronLopezGarcia

srobycap, 2025-03-23:
Hello, more information for the #528. The memory stick is 32 Go and I tried with a 4go too. A rollback to version 163 fixed the issue (with the same 32 Go memory stick with no filesystem modification) , I switched directly to this one because I read a post about it . Maybe the 164 or later are ok.
Sylvain.

JoseAaronLopezGarcia, 2025-03-23:
> Hello, more information for the [#528](https://github.com/PSP-Archive/ARK-4/issues/528). The memory stick is 32 Go and I tried with a 4go too. A rollback to version 163 fixed the issue (with the [...]

Was your issue correctly fixed by latest pre-release then?

srobycap, 2025-03-23:
I am curently reading all the answers. I used MacOS to communicate with the PSP (other information, It is my first use of ARK, I found my old PSP and wanted to revive it) At the begining I thought it was a corrupted filesystem, but not. I will try the lastest pre-release.

srobycap, 2025-03-23:
Hello,

the v1742755247 find here https://github.com/PSP-Archive/ARK-4/releases/tag/v1742755247 fixed the error. Everything seems ok now. 
Many thanks to you and schm1dtmac.

Sylvain

Rasegul, 2025-03-23:
Hey!
I‘m New to all this and somewhat a noob. I‘m experiencing the same issue. I have an old cso file from AGES ago that works just fine, but as I add any iso file (It seems) I just get that problem described above. I haven’t seen the ISOCACHE.Bin or something you guys mentioned before in the iso folder… Could you guys maybe water the steps down a bit? :)

srobycap, 2025-03-23:
It is well described into the README.txt which is find in the zip.

aayyubid, 2025-03-23:
Thanks. I appreciate it ^^