PSP-Archive/ARK-4 #529: Plugin Brightness Bug

FULANO93, 2025-03-23:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

Rev 167: UMDelay

### Homebrew/GAME (if applicable)

PS1 Games

### Plugins ( if applicable )

vsh, ms0:/seplugins/cxmb/cxmb.prx, on vsh, ms0:/seplugins/categorieslite/category_lite.prx, on vsh, ms0:/seplugins/dayviewer.prx, on vsh, ms0:/seplugins/xmbih.prx, on vsh, ms0:/seplugins/zerovsh_patcher.prx, off vsh, ms0:/seplugins/cfe/cfe_loader.prx, off all, ms0:/seplugins/brightness.prx, on all, ms0:/seplugins/bright3.prx, off all, ms0:/seplugins/noumd.prx, off all, ms0:/seplugins/npdrm_free.prx, off all, ms0:/seplugins/macrofire.prx, off all, ms0:/seplugins/prxshot_png_mod/prxshot_png_mod.prx, off all, ms0:/seplugins/prxshot++.prx, off all, ms0:/seplugins/prxshot.prx, off all, ms0:/seplugins/sssssjpg.prx, off all, ms0:/seplugins/ssssspng.prx, off all, ms0:/seplugins/sssssbmp.prx, off all, ms0:/CheatMaster/CheatMaster.prx, off game, ms0:/seplugins/cwcheat/cwcheat.prx, on pops, ms0:/seplugins/cwcheat/cwcheatpops.prx, on game, ms0:/seplugins/TempAR/tempar.prx, on pops, ms0:/seplugins/TempAR/tempar_lite.prx, off game, ms0:/seplugins/cushome/cushome.prx, on game, ms0:/seplugins/pspstates/states_menu_pict_mod.prx, off game, ms0:/seplugins/nploader.prx, on pops, ms0:/seplugins/popsloader/popsloader.prx, on pops, ms0:/seplugins/popsloader/cdda_enabler.prx, on ULUS10041 ULUS10160, ms0:/seplugins/cheatdevice_remastered.prx, off UCUS98653 UCUS98737 homebrew, ms0:/seplugins/brightness.prx, off UCUS98653 UCUS98737 homebrew, ms0:/seplugins/bright3.prx, on

### Steps to reproduce

I go into any PS1 game with the brightness plugin active and it doesn't work/activate

Well, I just updated from version 165 straight to version 167. Normally I just go into my games and start one... but this time I did it differently. I went into the advanced settings and went through all the fonts to see if they had fixed the problems with the letter _**Ç**_ in most of them. I did this because I saw that they had changed the fonts... so far everything was normal. I went back to the one I was already using and it is one of the 3 that has no bugs. In this case, the font used is **_CP885..._** and when I started a PS1 game, I saw that the brightness plugin wasn't working, but it worked normally on the XMB. I haven't tested it on PSP ISOs... I don't know if it's just in my head, but apparently the font in the menu of PS1 games seems to have changed too (this part might be in my head)...
A curious fact is that I tried changing the line
`all, ms0:/seplugins/brightness.prx, on`
for
`pops, ms0:/seplugins/brightness.prx, on`
and it still didn't work, but if you add the line at the end
`ps1, ms0:/seplugins/brightness.prx, on`
the plugin starts working again in PS1 games... another thing that seemed strange was the fact that when the brightness plugin didn't work in PS1 games, the bright3 plugin worked, but it's not even active for those games...

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

LRU

### Skip Sony Logos

All

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Off

### Disable UMD Drive

On

### Disable Analog Stick

Off

### Enable QA Flags

On

### Custom settings

# Luxor doesn't like inferno Cache
ULUS10201, infernocache, off
# Enable Extra RAM on GTA LCS and VCS
ULUS10041 ULUS10160 ULES00151 ULES00502, highmem, on

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-03-23:
Probably some issue in the PLUGINS.TXT file.

FULANO93, 2025-03-23:
I fixed it, it was written wrong because I passed the text file from the PSP to the cell phone and just used the cell phone's own HTML reader to copy, but in this case I always used it as is and it worked fine, I believe there was a bug when I was testing the fonts, at least that's what I believe happened.

[PLUGINS.TXT](https://github.com/user-attachments/files/19410831/PLUGINS.TXT)

FULANO93, 2025-03-23:
The ps1 home font looks differently to you too ?

[image]

JoseAaronLopezGarcia, 2025-03-23:
Yeah it does look different, but there's nothing anywhere in ARK capable of doing this, so it's certaintly an issue with your PLUGINS.TXT file or the configuration of one of your plugins.

You can use Vanilla Mode: hold start while launching the game to temporarily disable plugins and see if that changes anything.

FULANO93, 2025-03-23:
I tested holding down the start button when starting, but that way I didn't have access to the PS1 games, since they were in the categories... I copied one game just for testing and tried again, it continued the same way... I'm going to do other tests, such as reinstalling the current version, maybe reinstalling the version I was using previously... in my mind what might have caused this was actually the change in the fonts, since I tested them all until I went back to the one I was using... maybe the `all, ms0:/` or `pops, ms0:/` command didn't work as it should, I can't say, anyway, if it doesn't solve it, I'll replace or adding `always, ms0:/` or `ps1, ms0:/` to the end in order to get around this

FULANO93, 2025-03-23:
I reinstalled the latest version, the problem persisted... I went back to version 165 and it started working normally again. I don't know if the same problem occurs with version 166 because I never installed it

tested changing `all` to `always` and `pops` to `ps1` but the problem doesn't seem to be here, but adding `ps1, ms0:/seplugins/brightness.prx, on` to the end as a new line apparently gets around the problem

JoseAaronLopezGarcia, 2025-03-24:
This is potentially fixed in latest (pre)release.

FULANO93, 2025-03-24:
Tested and confirmed, working again as it should be...
Thank you all for the work you've done, I really appreciate it !

closed 2025-03-24 by JoseAaronLopezGarcia