PSP-Archive/ARK-4 #532: Several games not working on ARK-4

Thekyyubi, 2025-03-24:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

6.61 ARK 4.20.69 r167 ciPL

### Homebrew/GAME (if applicable)

ULUS-10273, ULUS-10188, ULUS-10171

### Steps to reproduce

Power Stone: Boot game and select game (Power Stone 1 or 2, doesn't matter, they both crash).
Crazy Taxi: Crashes when selecting any game (Crazy Taxi 1 or 2, they both crash).
Star Wars: Lethal Alliance: Crashes after loading.

### USB Charge

On

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

On

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

On

### Custom settings

always, usbcharge, on
always, overclock, on
always, wpa2, on
always, launcher, off
always, highmem, on
always, mscache, on
always, infernocache, off
always, disablepause, off
always, oldplugin, on
always, hibblock, on
always, skiplogos, off
always, hidepics, off
always, hidemac, on
always, hidedlc, off
always, noled, off
always, noumd, off
always, noanalog, off
always, qaflags, on
# Luxor doesn't like Inferno Cache
ULUS10201, infernocache, off
# Enable Extra RAM on GTA LCS and VCS
ULUS10041 ULUS10160 ULES00151 ULES00502, highmem, on

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

Thekyyubi, 2025-03-24:
I know that another user posted issues with Power Stone, and another user responded saying they were able to get Power Stone working, but did not explain how. I am having the exact same issue as the other poster.

JoseAaronLopezGarcia, 2025-03-24:
First of all, you're not on the latest release. Please update to latest release and try again.

Thekyyubi, 2025-03-24:
> First of all, you're not on the latest release. Please update to latest release and try again.

I have now updated to revision 168. Nothing has changed. All three games are still exhibiting the same issues, giving error code 80010016.

JoseAaronLopezGarcia, 2025-03-24:
I just tried Crazy Taxi Fare Wars right now and worked just fine.

I just noticed this in your bug report:

always, highmem, on

You have Extra RAM enabled, and many retail games don't like this. Try setting it back to "Auto" so it doesn't interfere where it shouldn't be.

Thekyyubi, 2025-03-24:
[image]

I have disabled highmem and reset the system. Still seeing a crash when selecting a game in Crazy Taxi.

JoseAaronLopezGarcia, 2025-03-24:
What is the format of these games? ISO or EBOOT.PBP?

Thekyyubi, 2025-03-24:
> What is the format of these games? ISO or EBOOT.PBP?

They are ISOs. I have downloaded several copies of the ISOs and tried them on my PSP 3000-- they work on the 3000 but do not work on the PSP Go.

JoseAaronLopezGarcia, 2025-03-24:
Oh so it's potentially a PSP Go only issue? I'll try it out on mine.

Thekyyubi, 2025-03-24:
> > What is the format of these games? ISO or EBOOT.PBP?
> 
> They are ISOs. I have downloaded several copies of the ISOs and tried them on my PSP 3000-- they work on the 3000 but do not work on the PSP Go.

Hold-- Converting to an EBOOT allowed me to actually select a game, but now the game hangs on loading.

JoseAaronLopezGarcia, 2025-03-24:
Yeah don't worry I'll test on my PSP Go and see if I can replicate.

JoseAaronLopezGarcia, 2025-03-24:
I just tested Power Stone on my PSP Go and had no issues at all.

What plugins do you have installed?

krazynez, 2025-03-24:
> I just tested Power Stone on my PSP Go and had no issues at all.
> 
> What plugins do you have installed?

Yep works fine for me also

krazynez, 2025-03-24:
Tested both Crazy Taxi 1 & 2 and both Powerstones on ef0 and ms0 can't reproduce.

closed 2025-03-24 by krazynez

Thekyyubi, 2025-03-25:
> I just tested Power Stone on my PSP Go and had no issues at all.
> 
> What plugins do you have installed?

Apologies for late response. My plugins are:

GAME: npdrm_free_mod.prx, enabled
VSH: cxmb.prx, disabled
VSH: category_lite.prx, disabled
VSH: npdrm_free.prx, enabled
GAME: npdrm_free.prx, enabled
XMB: category_lite.prx, disabled

Out of curiosity, are you all using EBOOTs or ISOs?

JoseAaronLopezGarcia, 2025-03-25:
You have npdrm_free and npdrm_free_mod both installed at the same time. Get rid of npdrm_free and stick to npdrm_free_mod.

We use both and both should work fine.

JoseAaronLopezGarcia, 2025-04-21:
We figured out the issue only happens when your PSP Go doesn't have a memory stick (internal memory only).
Currently we're trying to find where the issue is, but it seems to have been introduced somewhere after 4.20 release.
More information and updates on this thread: https://github.com/PSP-Archive/ARK-4/issues/556