PSP-Archive/ARK-4 #534: "Scanning stubs for kxploit..."

Aromerac, 2025-03-25:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

4

### Steps to reproduce

Copy ARK_01234 to PSP/SAVEDATA/
Copy ARK_Loader to PSP/GAME/
Launch ARK-4 Loader from "Games stored on the memory card"

I have used homebrew on all my PSP forever, however I've been trying to run ARK on this (brand new and perfectly functional) 1000 and no matter what version nor what method I've used to copy the files it always get stuck on "Scanning status for kxploit...", with R trigger + power being the only way to get out and effectively taking me back to 660 of course.
Has anyone ever had a similar challenge and if so could help me solve this? Thank you so much in advance

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-03-25:
Usually this can happen with broken/corrupted memory cards. But just to be sure, you are running 6.60 or 6.61 right?

JoseAaronLopezGarcia, 2025-03-25:
> > Usually this can happen with broken/corrupted memory cards. But just to be sure, you are running 6.60 or 6.61 right?
> 
> Sí, estoy en 660, e intenté con dos tarjetas diferentes, una micro SD con convertidor, la cual funciona sin problemas para todo lo demás, y una MSPD Sandisk de 4Go.

La verdad que está raro eso. Puede que falle una vez, pero que falle siempre no es normal.
De momento mi consejo es instalar ChonoSwitch y usarlo para reinstalar 6.60 (o mejor aún, 6.61, así tienes mejor cobertura de internet), de esa forma partimos de un firmware completamente limpio (como salido de fábrica).

Aromerac, 2025-03-25:
> La verdad que está raro eso. Puede que falle una vez, pero que falle siempre no es normal. De momento mi consejo es instalar ChonoSwitch y usarlo para reinstalar 6.60 (o mejor aún, 6.61, así tienes [...]

Acabo de intentar con mi tercera y última tarjeta y tampoco, así que haré lo que dijiste y espero que salga bien. Muchísimas gracias.

JoseAaronLopezGarcia, 2025-03-25:
> Acabo de intentar con mi tercera y última tarjeta y tampoco, así que haré lo que dijiste y espero que salga bien. Muchísimas gracias.

De todas formas me preocupa la idea de que 6.60 no funcione. Para ARK no hay diferencia alguna entre 6.60 y 6.61 así que no debería "funcionar en uno y no en otro", ARK literalmente no hace nada diferente ni hace distinción entre ambos.

Aromerac, 2025-03-25:
> De todas formas me preocupa la idea de que 6.60 no funcione. Para ARK no hay diferencia alguna entre 6.60 y 6.61 así que no debería "funcionar en uno y no en otro", ARK literalmente no hace nada [...]

Bueno... Chronoswitch considera mi firmware no soportado, y al actualizar al 661 me sale DADADADA... A ver qué clase de modificaciones le hicieron a la pobre PSP. Lo raro es que luce como nueva y los UMD funcionan.

JoseAaronLopezGarcia, 2025-03-25:
Vale, ese error lo he visto antes, y sé que tiene solución.
En esta parte de este video te explican la solución: https://youtu.be/9IwVPWAYQaQ?t=1341