PSP-Archive/ARK-4 #540: WPA2 Plugin not working on newest versions

gvaldebenit, 2025-04-01:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

https://github.com/PSP-Archive/ARK-4/commit/9d861713baf1c7aae84924b0ef6924896b4d3eb0

### Plugins ( if applicable )

PS1, ms0:/seplugins/ccda_enabler.prx, on
ULJM05472, ms0:/seplugins/divapatch/divapatch.prx, on
game, ms0:/seplugins/prxshot.prx, on

### Steps to reproduce

WPA2 plugin not work after updating to the latest pre-release using online ARK-4 updater from XMB. It worked fine on the previous ARK version (r165).
Tested on a PSP 1000 with r165, with similar plugins and same network environment, and is able to connect to internet as also to detect WPA2 SSID.

Steps to check the Issue

1. Open Internet Browser. 
2. Select a WPA2 SSID saved on the console
3. Wait while trying connecting 
4. After some minutes, it fails to connect to network.

Additional and more quick way to check

1. Open network settings
2. Open Infrastructure Mode
3. Add a new connection
4. Use Scan
5. All WPA2 SSID will be gray out and Security labeled as "Not compatible"

Tested my home network and also a Hotspot configured as WPA2, both gray on Network scan, but not grayed out on PSP 1000 with r165. Also, it works perfectly fine with No Security Hotspot

### USB Charge

On

### CPU Clock in Game

Overclock

### CPU Clock in XMB

Auto

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

On

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

krazynez, 2025-04-01:
We haven't thouch any wpa2 stuff. It is still working on all my PSP fine.

gvaldebenit, 2025-04-01:
Yes, I checked that nothing was touched in WPA-2 side, but could be something related to the xmb refactors and translations made?
Also, there is a way for Ark Full Install to downgrade? If can somehow run r165 on my PSP 3000, I can test if is hardware related or not.

krazynez, 2025-04-01:
Just put r165 ARK_01234 and run ARK Loader

gvaldebenit, 2025-04-01:
Ok, did the downgrade to r165, and now is able to detect and connect to WPA2 networks. Updated from revision to revision up to r168, and now is working perfectly fine. Wanted to did a quick test on latest pre-release version, and again, WPA2 is grayed out,

JoseAaronLopezGarcia, 2025-04-02:
Fixed in latest pre-release.

closed 2025-04-02 by JoseAaronLopezGarcia