PSP-Archive/ARK-4 #541: Custom launcher cIPL menu reverts to Latin every time it's opened

L0u7, 2025-04-01:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

ARK 4.20.69 r163

### Plugins ( if applicable )

game, ms0:/SEPLUGINS/dayviewer_game.prx, on
xmb, ms0:/SEPLUGINS/dayviewer.prx, on
xmb, ms0:/SEPLUGINS/categorieslite/category_lite.prx, on
game, ms0:/SEPLUGINS/PRXSHOT/PRXSHOT.PRX, on
xmb, ms0:/SEPLUGINS/XMBIH.PRX, on

### Steps to reproduce

1. Open Custom Launcher in Extras
2. Menus load in Latin language, (Games, Files, Network, CFW settings, Plugins, Settings, Exit)
3. Scroll over to settings and change language to English
4. Exit, restart and shutdown all yield same result which is upon re-entering the custom launcher, the menu language is in Latin
5. Any other settings I change inside the custom launcher like menu animation, font, etc, stays saved when re-opening the custom launcher

### USB Charge

On

### CPU Clock in Game

Overclock

### CPU Clock in XMB

Overclock

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

All

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

On

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-04-01:
Check the date.

closed 2025-04-01 by JoseAaronLopezGarcia

L0u7, 2025-04-01:
Did you mean to update? I updated to 168 just now and still have the same issue