PSP-Archive/ARK-4 #542: Anything related to The Camera accessory doersn't work

SerialKillerSlash, 2025-04-01:
i open the camera app in the xmb, it crashes the entire psp, both when the camera is plugged in and not inserted, and when i try out audio mechanica (latest version) it gives me an error that says: "Error starting usbbus driver", i hope this issue gets fixed soon enough, because i missed taking pics with that camera (it's the 450X variant btw).

SerialKillerSlash, 2025-04-01:
k lemme try to turn usb charge off, also yes i have the latest pre-release

JoseAaronLopezGarcia, 2025-04-01:
Anything?

SerialKillerSlash, 2025-04-01:
First, the release you gave me was before the latest release but i'm chill about it, second, nope still the same issue, i guess the problem is from the mini usb driver according to that audio mechanica error (damn this is more confusing than i thought)

JoseAaronLopezGarcia, 2025-04-02:
Other users have said that camera work fine for them with this patch. In any case, just disable USB charge and you should be ok.

closed 2025-04-02 by JoseAaronLopezGarcia

SerialKillerSlash, 2025-04-02:
ok quick update, i just noticed i had RJL plugin turned on, i turned it off and it finally worked, srry for not checking my plugins but now, problem solved