PSP-Archive/ARK-4 #550: [PSP] Pressing "Information" on an ISO in the video section will sometimes mount it

WindedHero, 2025-04-12:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

ARK 4.2.0.69 r170

### Steps to reproduce

1. Have more than one ISO in the ISO/VIDEO folder
2. Press triangle on one of them
3. Press X on "Information"
4. PSP will reboot with the ISO mounted. This occurs for certain ISOs and not others, and behavior is not consistent. 

### USB Charge

Off

### CPU Clock in Game

Overclock

### CPU Clock in XMB

Overclock

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

On

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-04-12:
I am aware of this bug and I have no intention of fixing it. The information you can see will be fake anyways.

closed 2025-04-12 by JoseAaronLopezGarcia

JoseAaronLopezGarcia, 2025-04-30:
Fixed in latest pre-release.