PSP-Archive/ARK-4 #551: Garbled Text in CJK Menus (r170) - Works Fine in r168

Tsubamei, 2025-04-14:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

R170

### Steps to reproduce

[image]

A text display issue in the Chinese, Japanese, and Korean menus of the program in version r170. In this version, the text within these menus appears as garbled characters.
This issue was not present in the previous version, r168, where the CJK menu text was displayed correctly.

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-04-15:
Let me know if latest pre-release fixes it (https://github.com/PSP-Archive/ARK-4/releases/tag/v1744723585)

Tsubamei, 2025-04-16:
After testing, the issue is fixed. Thank you.

closed 2025-04-16 by JoseAaronLopezGarcia