PSP-Archive/ARK-4 #552: Update OTA freezes PSP

XestoX, 2025-04-16:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

170

### Homebrew/GAME (if applicable)

OTA Update

### Plugins ( if applicable )

categorys_lite
cw cheat

### Steps to reproduce

Hey, 

Update via WiFi directly from psp 2000 and 3000 freezed my psp. PSP 1000 and N1000 worked fine. IS this a known bug or did I do anythign wrong? I simply connected OTA it downloads perfectly, it launches the update but freeze on Game Boot logo. PSP shuts down after that. when I relaunch the update app out of Game folder, same. but only on my slim and my brite. fat and go works great (street has no wifi xD)

### USB Charge

On

### CPU Clock in Game

Overclock

### CPU Clock in XMB

Overclock

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

On

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

On

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-04-16:
I'm unable to reproduce this. I'm gonna need more information about your setup.

JoseAaronLopezGarcia, 2025-04-16:
Fixed in latest pre-release. Thanks for the report.