PSP-Archive/ARK-4 #553: A few options in Custom Firmware Settings dissapear and the remaining ones get incorrect choices

Soto-Jnthan, 2025-04-18:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

r171

### Steps to reproduce

1. Have access to a PSP 3000 with no CFW.
2. Download ARK-4 files for the r171 release.
3. Move or copy ARK_01234 folder into /PSP/SAVEDATA/ folder.
4. Move or copy ARK_Loader, ARK_cIPL and ARK_Full_Installer folders into /PSP/GAME/ folder.
5. Run the following in sequential order: ARK_Loader > ARK_cIPL > ARK_Full_Installer.
6. Remove ARK_01234, ARK_Loader, ARK_cIPL and ARK_Full_Installer from the PSP.
7. Go to the Extras > Custom Firmware Settings and enable Hide DLCs and Disable UMD Drive
8. Fully shutdown the PSP
9. Go to Custom Firmware Settings after turning on the PSP
10. Observe the missing settings and settings with incorrect options (e.g. "Enable WPA2 support" and "Autoboot Launcher" have OverClock)

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

On

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-04-18:
What language do you have in your system?

Soto-Jnthan, 2025-04-18:
System Language is set to English

JoseAaronLopezGarcia, 2025-04-18:
I'm unable to reproduce this issue. Try reinstlling ARK-4, perhaps some file/s got corrupted.

Soto-Jnthan, 2025-04-18:
I made sure to restore the PSP to default settings and format the memory stick, and I am still able to reproduce this issue 100% of the time. Have you done the full installation and remove all ARK files from the PSP afterwards? It only occurs after doing that

JoseAaronLopezGarcia, 2025-04-18:
Oh I didn't notice that you mentioned removing ARK_01234 folder.

JoseAaronLopezGarcia, 2025-04-18:
Fixed in latest pre-release

closed 2025-04-18 by JoseAaronLopezGarcia