PSP-Archive/ARK-4 #556: Ark 4 powerstone collect and crazy taxi

Hookz1990, 2025-04-21:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

Latest

### Homebrew/GAME (if applicable)

Powerstone collection 

### Steps to reproduce

Power stone collection crashes when selecting either power stone 1 or 2 and sends you back to the xmb

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

krazynez, 2025-04-21:
[Duplicate](https://github.com/PSP-Archive/ARK-4/issues/532)

JoseAaronLopezGarcia, 2025-04-21:
> Powerstone collection and crazy taxi crash when choosing whichever game and sends you back to the xmb also there is a couple of bugs where the analog stops working and also when I press the home button on my psp go and select the pause game function nothing happens

First of all, take a deep breath and try to explain things better. In a single sentence you mention 3 different issues.
We can't help you if you can't even properly tell us what the problem is.

1 - I have Crazy Taxi on my PSP Go and have zero issues.
2 - The pause function is disabled because Inferno Cache is enabled (this is explained in the WIKI).
3 - If your analog is not working properly this is a hardware issue, not ARK related.

Hookz1990, 2025-04-21:
What are you settings and are using iso format of the game

JoseAaronLopezGarcia, 2025-04-21:
> What are you settings and are using iso format of the game <br> <br> On
> <br> <br> &gt; Powerstone collection and crazy taxi crash when choosing
> whichever game <br> &gt; and sends you back to the xmb also there is a
> couple of bugs where the <br> &gt; analog stops working and also when I
> press the home button on my psp go and <br> &gt; select the pause game
> function nothing happens <br> &gt; <br> &gt; First of all, take a deep
> breath and try to explain things better. In a <br> &gt; single sentence you
> mention 3 different issues. <br> &gt; We can&#39;t help you if you
> can&#39;t even properly tell us what the problem is. <br> &gt; <br> &gt; 1
> - I have Crazy Taxi on my PSP Go and have zero issues. <br> &gt; 2 - The
> pause function is disabled because Inferno Cache is enabled (this <br> &gt;
> is explained in the WIKI). <br> &gt; 3 - If your analog is not working
> properly this is a hardware issue, not <br> &gt; ARK related. <br> &gt;
> &gt; &lt;
> <br> &gt; <br> &gt; Powerstone collection and crazy taxi crash when
> choosing whichever game <br> &gt; and sends you back to the xmb also there
> is a couple of bugs where the <br> &gt; analog stops working and also when
> I press the home button on my psp go and <br> &gt; select the pause game

I'm using ISO format and default settings. Only tested with Internal Memory
as I don't have an M2 or adapter.

Hookz1990, 2025-04-21:
Yes I'm using iso format of the game and installed the game on the internal storage can you send me a list of your settings so I can give it a try also using  a psp go

JoseAaronLopezGarcia, 2025-04-21:
What plugins do you have installed?

JoseAaronLopezGarcia, 2025-04-21:
> When I go to view pluging it just says example on so don't think I'm using
> any at the moment

Are you trying to run this from the XMB or the Custom Launcher?

Hookz1990, 2025-04-21:
From the xmb and also tried custom launcher

JoseAaronLopezGarcia, 2025-04-21:
Try copying the game again. It really shouldn't have any issues.

Hookz1990, 2025-04-21:
OK do I need to change any settings under settings  also I will give it a whirl now

Hookz1990, 2025-04-21:
OK will give it a try now and let you know

Hookz1990, 2025-04-21:
Also should I launch game from xmb or custom launcher does it matter

Hookz1990, 2025-04-21:
Just tried launching the game both ways from xmb and custom launcher but still same error also is there a way to reset the firmware settings sorry I'm new to all this

JoseAaronLopezGarcia, 2025-04-21:
In Advanced VSH Menu.

Hookz1990, 2025-04-21:
Oh ok just reset the setting now I will give it another try now and let you know how I get on

Hookz1990, 2025-04-21:
No still no luck I'm running out of ideas it very strange al my other games work fine

JoseAaronLopezGarcia, 2025-04-21:
Which version of the game is it?

Hookz1990, 2025-04-21:
It is power stone collection Eu I downloaded it online

JoseAaronLopezGarcia, 2025-04-21:
And crazy taxi is also eu?

JoseAaronLopezGarcia, 2025-04-21:
> Yes I believe so

Try holding start when booting up the game.

Hookz1990, 2025-04-21:
I've just tried that still no luck

Hookz1990, 2025-04-21:
I'm just going to try and find another version of the rom now hopefully can solve the issue

krazynez, 2025-04-21:
> I'm just going to try and find another version of the rom now hopefully can solve the issue

Try the US version

JoseAaronLopezGarcia, 2025-04-21:
My version of Crazy Taxi is Europen too so I don't know if that has
anything to do with it.
If I could reproduce the issue it would be easy for me to figure out a
solution, but I can't do it blindly.

Hookz1990, 2025-04-21:
OK will do now

JoseAaronLopezGarcia, 2025-04-21:
What's your system language? I've seen games before that crash when you
select certain languages.

krazynez, 2025-04-21:
Also are you launching from internal storage ( ef0 ) or Memory Stick ( ms0 )

Just for clarification really. What I remember I tested on both in the other posted issue.

Hookz1990, 2025-04-21:
System language is English
And launching from internal storage as I don't have an sd card yet

JoseAaronLopezGarcia, 2025-04-21:
Everything looks good. Try downloading the game again, or another website.

JoseAaronLopezGarcia, 2025-04-21:
Managed to reproduce the issue. It happens when you don't have a memory stick installed.

Hookz1990, 2025-04-21:
OK so do I need to purchase a memory stick then for the game to work

krazynez, 2025-04-21:
> OK so do I need to purchase a memory stick then for the game to work

No we will work on fixing. Unless you really need to play it right now 🤣

Hookz1990, 2025-04-21:
OK thats excellent news thank you so much for your time and help

JoseAaronLopezGarcia, 2025-04-21:
Latest pre-release fixes the issue. Let me know if you find any other problems.

closed 2025-04-21 by JoseAaronLopezGarcia

Hookz1990, 2025-04-21:
How do I download  the pre release I already have ark for installed do I just do the update on my psp go sorry I'm new to all this

krazynez, 2025-04-21:
> How do I download  the pre release I already have ark for installed do I just do the update on my psp go sorry I'm new to all this

Just update its in stable now

Hookz1990, 2025-04-21:
Thank you so much I will definitely let you know if there's anymore issues keep up the amazing work guys