PSP-Archive/ARK-4 #557: Per-game plugins not working

CarnageDevs, 2025-04-22:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

r173 (Eli)

### Homebrew/GAME (if applicable)

ULUS10160

### Plugins ( if applicable )

"xmb, ms0:/seplugins/CXMB/CXMB.prx on"
"xmb, ms0:/seplugins/category/category_lite.prx on"
"always, ms0:/seplugins/PspPowerSaver.prx on"
"always, ms0:/seplugins/prxshot_png_mod/prxshot_png_mod.prx on"
"game, ms0:/seplugins/CUSHOME/CUSHOME.prx on"
"ULUS10160, ms0:/seplugins/cheatdevice_remastered.prx on"

### Steps to reproduce

1. Follow the wiki steps to enable per game plugins for Cheat Device Remastered for GTA VCS/LCS (or both), which will lead to a line like this in your PLUGINS.txt: `ULUS10160, ms0:/seplugins/cheatdevice_remastered.prx on`
2. Launch GTA VCS/LCS and press L + UP and nothing happens

I can confirm this is not misconfiguration on my part as if I enable the plugin for all games with this line:
`game, ms0:/seplugins/cheatdevice_remastered.prx on`

The cheat device menu works as expected and is openable in-game. I can also verify that this affects the disabling of per game plugins as well, as if I edited my PLUGINS.txt to enable cheat device for all games and tried to disable it for GTA VCS/LCS specifically:
```
game, ms0:/seplugins/cheatdevice_remastered.prx on
ULUS10160, ms0:/seplugins/cheatdevice_remastered.prx off
```

It still works in game. The issue cannot be related to the game IDs either as I grabbed them straight from the custom launcher as per wiki instructions, and they were correct as I'm playing the US versions of the games.

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Forced

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-04-22:
Yes I am aware of this, but didn't have time yesterday to fix it. Will check it out as soon as I can.
Thanks for the report.

JoseAaronLopezGarcia, 2025-04-22:
Fixed in latest pre-release.