PSP-Archive/ARK-4 #559: Custom Launcher freezing then psp switching off

Karim0786, 2025-04-22:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.69 r174

### Homebrew/GAME (if applicable)

VSH

### Plugins ( if applicable )

[PLUGINS.TXT](https://github.com/user-attachments/files/19855261/PLUGINS.TXT)

[PLUGINS.TXT](https://github.com/user-attachments/files/19855265/PLUGINS.TXT)

### Steps to reproduce

i do not know how
I guess fix the freezing

### USB Charge

On

### CPU Clock in Game

Overclock

### CPU Clock in XMB

Overclock

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Custom settings

[SETTINGS.TXT](https://github.com/user-attachments/files/19855311/SETTINGS.TXT)

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-04-22:
We need a little more information than that.

JoseAaronLopezGarcia, 2025-04-23:
Try disabling all plugins to see if that's the issue.

closed 2025-04-24 by JoseAaronLopezGarcia