PSP-Archive/ARK-4 #560: OTA Updater will not update to latest release from penultimate revision (and it does from earlier ones)

Soto-Jnthan, 2025-04-22:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

r173

### Steps to reproduce

1. Have access to a PSP 3000 with no CFW.
2. Download ARK-4 files for the rev173 release.
3. Move or copy ARK_01234 folder into /PSP/SAVEDATA/ folder.
4. Move or copy ARK_Loader folder into /PSP/GAME/ folder.
5. Run ARK_Loader.
6. Go to Settings > ARK-4 Updater
7. Choose "Update via Internet" by connecting to a nearby wireless hotspot
8. Observe the prompt claiming the latest version has been already installed (Note that as of writing, rev174 is this latest version; this is important in case the bug is related to an Off-by-One comparison)

### USB Charge

On

### CPU Clock in Game

Overclock

### CPU Clock in XMB

Overclock

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

On

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

krazynez, 2025-04-22:
You needed to wait for me to update the actual server 😉

closed 2025-04-22 by krazynez

Soto-Jnthan, 2025-04-22:
> You needed to wait for me to update the actual server 😉

Oh, and why did the server work from rev172 to rev174 and not from rev173 to rev174?

krazynez, 2025-04-22:
> > You needed to wait for me to update the actual server 😉
> 
> Oh, and why did the server work from rev172 to rev174 and not from rev173 to rev174?

Did you use /test that would have went to 174 before me updating stable

Soto-Jnthan, 2025-04-22:
> > > You needed to wait for me to update the actual server 😉
> > 
> > 
> > Oh, and why did the server work from rev172 to rev174 and not from rev173 to rev174?
> 
> Did you use /test that would have went to 174 before me updating stable

Sorry, I do not get your question. But I tested it when I opened the issue. I went from r172 to r174 with no issues and could not go from r173 to 174 for some reason. Intially I thought it was related to the update still not being visible from the server side, but that does not make much sense when I was able to update from r172 to r174 using the same server.