PSP-Archive/ARK-4 #561: [PSP] Always launch PS1 games in 480i

triple-lei, 2025-04-24:
### New Feature

I play my PSP games on a TV with component cables and a video scaler (e.g. RetroTINK 4K). The scaler accepts 480p and I'm able to zoom the normally windowboxed screen to play PSP games in full screen. So far, so good. 

The PSP can also play PS1 games in 480p using component cables, but it's just better to display PS1 games in their native 240p and let the scaler cleanly upscale to 1080p. You do this by setting the PSP's Connected Display Settings to "TV Type: 4:3" and "Component / D-Terminal Output: Interlace." 

Problem is, I have to scroll through the XMB just to toggle those settings every time I want to play a PS1 game, and then reverse it when I want to play a PSP game. Worse, if I don't bother changing settings, the scaler will still see a zoomed-in 480p picture and I have to undo that as well before I can play the PS1 game. Ideally, ARK-4 firmware would recognize that a PS1 game is being played and automatically set the PSP to 4:3 / Interlaced, and then switch back to 16:9 / Progressive when I'm back in the XMB. Or if that's not possible, even a one-touch toggle in the VSH Menu would be a great timesaver.

JoseAaronLopezGarcia, 2025-04-24:
Have you tried the FusaSD plugin? I'm not entirely sure if this must be done by the CFW or by a plugin.

triple-lei, 2025-04-26:
Just tried FuSa on the latest ARK-4 right now and I get a freeze at the white PSP splash screen when starting a PSP game. So I guess that's that!

And anyway, I was under the impression that (1) you don't want to load too many plugins at a time, and (2) video output plugins in particular were kinda iffy. I remember back in the early days of YouTube, there were tons of videos showing games running slow, all recorded with RemoteJoyLite and PC screen recording software. Overclocking should solve this but you can never be too sure. I know FuSa mentions a SpeedDBoosteR mode, but I'd rather steer clear of that altogether.

The less I have to think about settings, the better! Hence my request of auto-switching to 480i when playing PS1 games. Just seemed like an elegant solution, not that I know the first thing about creating CFW or homebrew myself.

JoseAaronLopezGarcia, 2025-04-26:
Ah I see, FusaSD is indeed kinda iffy and it apparently doesn't work on 6.60 anyways. I'll see if something can be cooked up.