PSP-Archive/ARK-4 #562: I may just be dumb and I'm sorry if that's the case.

Duothedouble, 2025-04-27:
Plugins.txt
xmb, ms0:/seplugins/categorieslite/category_lite.prx, on
xmb, ms0:/seplugins/cxmb/cxmb.prx, off

I just want my games to show up on the games list and they don't I don't know what I did wrong but I can only access my .iso's from the file list. Is there anyone who can explain to me what I did wrong.

JoseAaronLopezGarcia, 2025-04-27:
ISOs go in the /ISO/ folder, others go in the /PSP/GAME/ folder. This isn't ARK related.
Check online for more instructions related to category lite.

closed 2025-04-27 by JoseAaronLopezGarcia

Duothedouble, 2025-04-29:
I'm sorry I'm new to the scene, ( Haven't had a psp since i was in middle school ) In my file system I don't have an /ISO file and I'm having trouble finding anything about category lite, I thought it might be ark that I might have messed up the installation. Sorry for asking for help in the wrong place just haven't been able to find anything that explains what I'm supposed to do I followed the guide on pspunk to mod my psp and the result wasn't what was advertised.. again I may just be dumb and overlooked something will run thru the steps again and see if I get a better result.