PSP-Archive/ARK-4 #568: Chinese game titles are not supported.

cooto999, 2025-05-02:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

ARK-4-v1746136261

### Steps to reproduce

I downloaded some PSP ISO games with Chinese filenames, such as "GT赛车.iso," but when I put them on the memory card and try to run them, I get error ​​80010016​​. However, if I rename them to English, they work fine.
Please fix it.
Thanks.

### USB Charge

On

### CPU Clock in Game

Overclock

### CPU Clock in XMB

Overclock

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

RR

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

cooto999, 2025-05-02:
> XMB or Custom Launcher? Have you tried both?

both

JoseAaronLopezGarcia, 2025-05-02:
I'm going to asume that the system language is also Chinese right?
I have tested the ability to launch ISOs with non-latin characters before, but maybe something broke it.

cooto999, 2025-05-02:
> I'm going to asume that the system language is also Chinese right? I have tested the ability to launch ISOs with non-latin characters before, but maybe something broke it.

Yes, the system is in Chinese. Additionally, even if I switch the system language to English, games with Chinese filenames still cause the same issue.

JoseAaronLopezGarcia, 2025-05-03:
I am unable to reproduce this issue. I have no problem loading games with chinese characters in the file name (I even used the sample you provided and it worked just fine).
Make sure the ISO itself is not corrupted. It could also be an issue with the memory stick or sd adpater.

mo10, 2025-06-02:
I encountered a similar issue, but the error code was 800200D9.

Here's what I tried:

- I renamed the non-working Chinese name ISO to an English name, and it worked fine.
- I renamed other working ISO to the same Chinese name, and it worked fine.

I enabled the bsod.prx plugin, but it didn't seem to capture any exceptions.

I'm a bit confused, it appears the issue only occurs with specific content with specific names.
I've uploaded the game here, hoping it will help reproduce the issue.
https://mega.nz/file/4IkxwKyQ#DG0EsET1Hdkj93nxhuwRAArCZPGRLmuZG3sVSzkUlX8

kurikomoe, 2025-07-06:
I met this too:
fresh installed ark4 (temporary installation) + fully reset psp 3000(9g) + no plugins

with the name `女神异闻录3 携带版 v1.1[简][扑家汉化组].iso`, error code 80010016.
renamed to "persona3.iso", the game starts and works fine.

kurikomoe, 2025-07-06:
> Are you sure its an error related to the Chinese characters and not the length of the file name?

I am pretty sure that this file works previously on infinity cfw. 
I just switched from infinity cfw to ark4 today. Copied back the games, and found that nearly all my games fails to start (all in Chinese filenames).

mo10, 2025-07-06:
> I met this too: fresh installed ark4 (temporary installation) + fully reset psp 3000(9g) + no plugins
> 
> with the name `女神异闻录3 携带版 v1.1[简][扑家汉化组].iso`, error code 80010016. renamed to "persona3.iso", the game starts and works fine.

PSP-3000 6.61 with ARK 4.20.69 r180 + cIPL

Tried this name with XMB launch, successs reproduced this error.
It appears as an extended black screen on the Custom Launcher, followed by a full system restart. But no error prompt happened.

JoseAaronLopezGarcia, 2025-07-06:
Alright I will look into it.

JoseAaronLopezGarcia, 2025-07-07:
Hello, I am completely incapable of reproducing this issue using the latest release and pre-release.

Here's what I did:

- On a PSP Go I renamed some random ISO to "GT赛车.iso" and it booted just fine from both the XMB as well as Custom Launcher.
- On a PSP 3000 I renamed some random ISO to "女神异闻录3 携带版 v1.1[简][扑家汉化组].iso" and it booted just fine from both the XMB as well as Custom Launcher.

This one is a puzzle, specially since ARK really just does the same patches that PRO and ME do to enable non-latin characters in ISO file names, and it's working fine on my end.

Perhaps there's some plugin that causing some incompatibility? Can you try launching the ISO (which chinese characters) while holding the Start button? That should temporarily disable plugins.

mo10, 2025-07-07:
> What about plugins? Have you tried the latest pre-release?

1. Tried https://github.com/PSP-Archive/ARK-4/releases/tag/v1751878954
2. Copy UPDATE to /PSP/GAME
3. Copy  ARK_01234 to PSP/SAVEDATA
5. Plugins: leda.prx ON; bsod.prx OFF; No other plugins 

Still popping up error. 
I am trying some different options, like memory stick overclocking and inferno cache...

kurikomoe, 2025-07-07:
> > Do they also fail to load with Custom Launcher?
> 
> Here's some more tests:
> 
>     1. rename from "persona3.iso" to "女神异闻录3.iso" => failed to launch with 80010016
> 
>     2. use custom launcher + "女神异闻录3.iso" => failed to launch with 80010016
> 
>     3. in usb mode, rename the file back to "persona3.iso", directly launch => works
> 
> 
> > Feel free to request additional test results or other information.
> 
> Here is the sample file, though I think all games named in Chinese won't work.
> 
> ENV: PC: Windows in English but with GBK locale. PSP: all reset + ark4 installed in temporary mode, no plugins, default settings. PSP system: 6.61 + ark(r187) TF Card formatted by PSP.
> <img alt="Image" width="690" height="883" src="https://private-user-images.githubusercontent.com/31176859/462910862-9cda80c3-a9a8-4c26-9c6d-cf78c20f55eb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTE4OTA4NjQsIm5iZiI6MTc1MTg5MDU2NCwicGF0aCI6Ii8zMTE3Njg1OS80NjI5MTA4NjItOWNkYTgwYzMtYTlhOC00YzI2LTljNmQtY2Y3OGMyMGY1NWViLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzA3VDEyMTYwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU4OTEyNWUwYjk0NDAxZmJhM2NkYjc3OGI1OWY4NjY0MTYyZWZhYzc0YjA1YjBiOWIyNGY3ZDdiNWYxZjUwZDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.3VQwcKEcmLYuWCet8mCxX5cM6FmBfI_LVDuXNyui4Eg">

I remembered that I have met the situation: when the Windows locale changed to Unicode, some applications will fail on accessing the FAT32 driver. So I guess the locale indeed impacts the filename encoding or other things on FAT32.

JoseAaronLopezGarcia, 2025-07-07:
But why was it working on PRO?
I wonder if there's a way we can work around this.

JoseAaronLopezGarcia, 2025-07-07:
Well ARK uses the same code that PRO uses to allow non-latin characters.
This does indeed seems to be an issue with Windows, as Linux uses UTF-8 and that has no problem with the PSP that also expects UTF-8, but windows probably uses CP936 or some other encoding made by Microsoft.

JoseAaronLopezGarcia, 2025-07-07:
@kurikomoe let me know if the latest pre-release helps, it certaintly did for @mo10

JoseAaronLopezGarcia, 2025-07-07:
You too pal. Take care.