PSP-Archive/ARK-4 #569: PSP GO internal storage not accessible with UMD movie mounted

pcalkovsky, 2025-05-02:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.69 r175

### Plugins ( if applicable )

No plugins 

### Steps to reproduce

1. choose UMD movie from memory stick
2. Psp reboots and shows movie as movie present in UMD drive. Movie can be played
3. Psp go internal storage completely disappears and cannot be used until the console is rebooted.

### USB Charge

On

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-05-02:
How is this a bug?

PSP Go has no UMD capabilities, XMB needs to be tricked into thinking that it's not running on a Go to allow for UMD capabilities, so of course Internal Memory goes away.

Be thankful that we can bring UMD capabilities to PSP Go that easily. If Sony wanted to, it could have been made impossible (look at PS Vita).

pcalkovsky, 2025-05-02:
Don't get me wrong I am definitely thankful. I just wasn't sure if it was bug or intended behaviour. I wanted to report it in case it was something that could be "fixed"