PSP-Archive/ARK-4 #570: wifi usb dongle support

Kostadinosk, 2025-05-04:
### New Feature

basically what the title says. wifi usb dongle support for the psp street because it doesn't have built in wifi.a

isage, 2025-05-04:
you can't connect normal usb devices to psp. it's hardware doesn't support it.

Kostadinosk, 2025-05-04:
Then why can I connect the camera and the gps

isage, 2025-05-04:
> Then why can I connect the camera and the gps

what exactly about "normal usb devices" isn't clear?

Camera and gps arent't normal usb devices. They act as usb host (like pc). Psp acts as client (dongle)

Kostadinosk, 2025-05-04:
So theoretically I could mod the firmware of a USB dongle for it to work

isage, 2025-05-04:
No...
You can't just mod fw of something and magically modify it's hardware.

Kostadinosk, 2025-05-04:
I mean change the firmware of the dongle to act as the host

isage, 2025-05-04:
ability to act like host (or switch between client/host) is a hardware feature, not software/firmware.

Kostadinosk, 2025-05-04:
Ok if I buy one that acts like a host

isage, 2025-05-04:
...
There's zero usb dongles that can act like host, because that doesn't make any sense.
The only mass-market (except some RPi) devices that can act as both client and host are mobile phones.

Kostadinosk, 2025-05-04:
Ok so if I do it with a raspberry pi Pico W will I have a wifi connection then

Double-0-seven7, 2025-05-04:
You could probably tunnel the USB connection to infrastructure play somehow.
You could already do this for adhoc (xlinkkai) with the cwusb/adhoctousb plugin but for some reason it doesn't work on the psp street...

Kostadinosk, 2025-05-04:
For the only device that doesn't have a wifi connection

Kostadinosk, 2025-05-04:
> You could probably tunnel the USB connection to infrastructure play somehow when its connected to the PC. You could already do this for adhoc (xlinkkai) with the cwusb/adhoctousb plugin but for some reason it doesn't work on the psp street...

Also I don't wanna connect for multiplayer I wanna connect for browser and online stuff

isage, 2025-05-04:
> Ok so if I do it with a raspberry pi Pico W will I have a wifi connection then

Sure, if you
1) make hardware to allow it to be powered from psp (psp doesn't provide power to devices over usb)
2) make software for PI that allows it to act as wifi dongle, with usb host physically, but usb client logically 
3) make driver for psp to allow it to a) communicate with such device b) route network calls through such device

Kostadinosk, 2025-05-04:
> Sure, if you
>
> 1. make hardware to allow it to be powered from psp (psp doesn't provide power to devices over usb)
> 2. make software for PI that allows it to act as wifi dongle, with usb host [...]

Wait so for the 1 how is the PSP using the gps module and the camera they don't have batteries in them right?

isage, 2025-05-04:
> Wait so for the 1 how is the PSP using the gps module and the camera they don't have batteries in them right?

It provides power via prongs outside of port.
It's also 3v3, not 5v like most usb devices

Kostadinosk, 2025-05-04:
Sony be like the Nintendo of the time

JoseAaronLopezGarcia, 2025-05-04:
Even if this were possible, I don't see why it needs to be part of the CFW instead of a separate plugin.

closed 2025-05-04 by JoseAaronLopezGarcia