PSP-Archive/ARK-4 #575: Test

RandomGuy2023, 2025-05-08:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

4.20.69 r179

### Steps to reproduce

When clicking a setting (like "Games", "Network", etc.) after clicking the "Triangle" button right after running PS Vita's standalone ARK-4 when it's loading PSP games, ARK-4 freezes and the PS Vita hangs when turning it off or after closing ARK-4 and running it again.

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-05-09:
So it only happens when it's loading games? I'll check it out.

RandomGuy2023, 2025-05-12:
> So it only happens when it's loading games? I'll check it out.

I also can't access PSP's ARK-4's Vanilla Mode.

JoseAaronLopezGarcia, 2025-05-12:
> > So it only happens when it's loading games? I'll check it out.
> 
> I also can't access PSP's ARK-4's Vanilla Mode.

What do you mean you can't access Vanilla Mode? That's not something you "access", it's something you enable temporarily.

RandomGuy2023, 2025-05-12:
> > > So it only happens when it's loading games? I'll check it out.
> > 
> > 
> > I also can't access PSP's ARK-4's Vanilla Mode.
> 
> What do you mean you can't access Vanilla Mode? That's not something you "access", it's something you enable temporarily.

I mean, after holding SELECT and/or START during bootup or gameboot, the XMB keeps the same and the settings and plugins remain enabled.

JoseAaronLopezGarcia, 2025-05-12:
> 
> the settings and plugins remain enabled.

What exactly do you mean by this? If you mean that SETTINGS.TXT and PLUGINS.TXT are kept the same, then:

Let me say it once again: Vanilla Mode disables settings and plugins **_TEMPORARILY_**, if you were expecting the entirety of SETTINGS.TXT and PLUGINS.TXT to be modified so all settings and plugins are disabled then that would be **_PERMANENT_**, not **_TEMPORARY_**.

On a side note: closing this issue as it should be fixed in latest pre-release.

closed 2025-05-12 by JoseAaronLopezGarcia

RandomGuy2023, 2025-05-12:
> What exactly do you mean by this? If you mean that SETTINGS.TXT and PLUGINS.TXT are kept the same, then:
>
> Let me say it once again: Vanilla Mode disables settings and plugins **_TEMPORARILY_**, if [...]

No, I mean, none of the settings are changed, even temporarily.

JoseAaronLopezGarcia, 2025-05-12:
> No, I mean, none of the settings are changed, even temporarily.

Which settings? Everything is working fine here.

RandomGuy2023, 2025-05-12:
> Which settings? Everything is working fine here.

"Custom Firmware Settings", "Plugins Manager" and "Custom Launcher" under "Extras" on the XMB.

Also, XMB and Custom Launcher/PS Vita's/PS TV's standalone ARK-4 still often don't detect a new ARK-4 version when it was just released and I have to wait some time for them to do it or manually install it. Moreover, when scrolling games on PS Vita's/PS TV's standalone ARK-4, the consoles still make noise even with no sound.

JoseAaronLopezGarcia, 2025-05-12:
> 
> "Custom Firmware Settings", "Plugins Manager" and "Custom Launcher" under "Extras" on the XMB.
> 

Those aren't settings.

>
> Also, XMB and Custom Launcher/PS Vita's/PS TV's standalone ARK-4 still often don't detect a new ARK-4 version when it was just released and I have to wait some time for them to do it or manually install it.

The stable release server has to be manually updated and the person in charge of the server isn't always available exactly on release.

>
>Moreover, when scrolling games on PS Vita's/PS TV's standalone ARK-4, the consoles still make noise even with no sound.

Mute works fine for everyone else. Check your taihen plugins, it could be any one of them.

RandomGuy2023, 2025-05-12:
> Those aren't settings.
>
>
> The stable release server has to be manually updated and the person in charge of the server isn't always available exactly on release.
>
>
> Mute works fine for everyone else. [...]

Then could you give me an example which settings are affected?

I didn't try mute, only with no sound. Anyway, even if it works, it doesn't make sense to do it just to run ARK-4, furthermore if playing a game afterwards with sound (since one would have to unmute it). Already did and still have this issue.

JoseAaronLopezGarcia, 2025-05-12:
> Then could you give me examples of settings that are affected?
> 

https://github.com/PSP-Archive/ARK-4/wiki/Custom-Firmware-Settings

> I didn't try mute, only with no sound.

Isn't "mute" and "no sound" the same thing? At least the dictionary definition of "mute" says so.
In any case, ARK is a PSP homebrew that has no control over the PS Vita's audio hardware, so ARK cannot possibly tell the PS Vita to emit audio when the Vita itself has decided to have no audio, that's for the Vita to decide. This is surely some native Vita-side taihen plugin.

RandomGuy2023, 2025-05-12:
> https://github.com/PSP-Archive/ARK-4/wiki/Custom-Firmware-Settings
>
>
> Isn't "mute" and "no sound" the same thing? At least the dictionary definition of "mute" says so. In any case, ARK is a PSP [...]

Not exactly in this context. Anyway, I also tried muting it now and I had the same issue. I don't think so. I also have it on the PS TV which has even less plugins and I tried on both consoles with only the necessariy plugins for HENkaku and ARK-4 and I had this issue.

JoseAaronLopezGarcia, 2025-05-12:
> Not exactly in this context. Anyway, I also tried muting it now and I had the same issue. I don't think so. I also have it on the PS TV which has even less plugins and I tried on both consoles with only the necessary plugins for HENkaku and ARK-4 and I had this issue.

Wait hold on a second, how exactly are you having this issue on a Vita TV when on that device the volume is controlled by the TV and not the Vita? None of what you are saying makes any sense.

RandomGuy2023, 2025-05-12:
> Wait hold on a second, how exactly are you having this issue on a Vita TV when on that device the volume is controlled by the TV and not the Vita? None of what you are saying makes any sense.

Forget what I said regarding the PS TV. I made a confusion with the previous crashing issue. I only have this issue on the PS Vita, which is odd enough.

RandomGuy2023, 2025-05-12:
> > Then could you give me examples of settings that are affected?
> 
> https://github.com/PSP-Archive/ARK-4/wiki/Custom-Firmware-Settings
> 

But those were exactly the settings I was talking about. They remain unchanged in "Vanilla Mode".

JoseAaronLopezGarcia, 2025-05-12:
> > > Then could you give me examples of settings that are affected?
> > 
> > 
> > https://github.com/PSP-Archive/ARK-4/wiki/Custom-Firmware-Settings
> 
> But those were exactly the settings I was talking about. They remain unchanged in "Vanilla Mode".

Please go check on a dictionary the meaning of the word **_TEMPORARY_** and the difference it has with the word **_PERMANENT_**.

I will no longer reply to this thread as all the questions you've asked so far can be easily answered with simple google searches or by reading the wiki.

RandomGuy2023, 2025-05-13:
> Please go check on a dictionary the meaning of the word **_TEMPORARY_** and the difference it has with the word **_PERMANENT_**.
>
> I will no longer reply to this thread as all the questions you've [...]

Like I said, the settings DON'T EVEN TEMPORARILY CHANGE.

No, they can't, hence why it isn't told why ARK-4 doesn't detect a new release even after it was released and why the information regarding Vanilla Mode is wrong, not to mention it doesn't say why it makes the PS

RandomGuy2023, 2025-06-10:
> What exactly do you mean by this? If you mean that SETTINGS.TXT and PLUGINS.TXT are kept the same, then:
>
> Let me say it once again: Vanilla Mode disables settings and plugins **_TEMPORARILY_**, if [...]

I tried the latest pre-release since you're taking a lot of time to release a new stable one and even though the consoles no longer freeze, I have some delay when clicking the settings.

JoseAaronLopezGarcia, 2025-06-10:
The PSP is a device from 2004, you can't expect instant load times like on the PS5.

JoseAaronLopezGarcia, 2025-06-11:
> Didn't you say you'd no longer reply to this thread?
> 

Who the hell are you to tell me what I can or cannot do? I'll reply to anything and everything if I feel like doing so.

> PSP? I was talking about the PS Vita and the PS TV.
> 

If you are talking about Native Vita stuff then this isn't the place to do so, this is a PSP-related software project.

> Anyway, when will the next stable release be available with this issue fixed?

I don't know, the issue is non-critical.

> Also, what about the sound and Vanilla Mode issues?

Sound issues? Never heard of them. Nobody else has ever talked about this. So it must be a YOU issue.

Vanilla Mode has no issues either, it's YOU who can't understand what Vanilla Mode does.
I can't be held responsible for you not understanding that Vanilla Mode is TEMPORARY and does NOT touch any text files related to settings or plugins. Please check with a dictionary for the meaning of TEMPORARY.

RandomGuy2023, 2025-06-12:
> just to clarify this issue happens to me too but its fixed by waiting for games to load which only takes 5sec

Hence the delay, like I said, and a stable release with it hasn't been released yet, not to mention I still have the other 2 issues on the latest pre-release.

RandomGuy2023, 2025-06-12:
> > > just to clarify this issue happens to me too but its fixed by waiting for games to load which only takes 5sec
> > 
> > 
> > Hence the delay, like I said, and a stable release with it hasn't been released yet, not to mention I still have the other 2 issues on the latest pre-release.
> 
> Well, being an entitled prick to a contributor wont get this done any faster so chill

I think you're confusing who's being a prick and it seems it won't get fixed, anyway.

JoseAaronLopezGarcia, 2025-06-12:
> 
> Well, being an entitled prick to a contributor wont get this done any faster so chill

Specially when I keep explaining to him that what he thinks is a bug is in fact not one at all.

> 
> I think you're confusing who's being a prick and it seems the issues won't be fixed, anyway.

You are being a prick when I keep trying to explain to you how things work and you keep saying that "no that's a bug".
The plugins and settings manager show currently configured plugins/settings, not currently running. CONFIGURED is not the same as RUNNING. Vanilla Mode prevents plugins and settings from being enabled, but it does't modify the text files where plugins and settings are configured, otherwise if it did then it would be PERMANENT as oppose to TEMPORARY.

A fucking 5 year old should be able to understand this, I don't know why you keep brining it up.
If English is not your main language then there's not much I can do here, you are literally the only person on earth that hasn't been able to understand how Vanilla Mode works, absolutely nobody else (not even the person who requested this feature years ago) has ever had any issue understanding what it does and doesn't do.

You also mention some "sound issue" that apparently only YOU have, every other human being on earth that's tried ARK has said that if anything ARK fixes sound issues in some PS1 games. Nobody has ever complained about any sound issues at all.

RandomGuy2023, 2025-06-12:
> Specially when I keep explaining to him that what he thinks is a bug is in fact not one at all.
>
>
> You are being a prick when I keep trying to explain to you how things work and you keep saying that [...]

No, you didn't. You thought I was talking about the PSP when I was referring to the PS Vita. It might be hardware-related and if so there's nothing to do about it but it happens on the PS Vita, not on the PSP like you said.

You are being a prick because you're lying since I already told you Vanilla Mode doesn't prevent plugins and settings from being enabled in my case so it doesn't even do TEMPORARY changes.

I guess a fucking 5 year old would get that he was told Vanilla Mode doesn't even TEMPORARILY prevent plugins and settings from being enabled but he also wouldn't fulfill his promise of no longer replying to this thread like you did. I also think he'd understand the issue which MAKES THE CONSOLE FREEZE is indeed CRITICAL so a stable release with it fixed should be out (even if it had the delay I complained about, regardless if there's nothing to do about it since it might be hardware-related).

My friends also have the sound issue but I bet if someone created a thread regarding it or mentioned it in this one, you'd accuse them of being me.

JoseAaronLopezGarcia, 2025-06-16:
> I guess a fucking 5 year old would get that he was told Vanilla Mode doesn't even TEMPORARILY prevent plugins and settings from being enabled but he also wouldn't fulfill his promise of no longer [...]

Abolutely NOBODY has any issues with Vanilla Mode. LITERALLY ZERO PEOPLE.
The only reason YOU think it's broken is because Vanilla Mode doesn't change the text files so you keep seeing enabled plugins in Plugin Manager, but if you had a single funcitioning brain cell you'd realize that it's also true for plugins that aren't even supposed to load on the XMB. That's because the Plugin Manager is not showing currently LOADED plugins, it's showing currently INSTALLED plugins. A plugin being installed doesn't mean it'll be loaded, specially if the plugin isn't even supposed to load on the XMB.
If you install (e.g.) PopsLoader, you will see the plugin as ENABLED despite the fact that you are in the XMB, not on a PS1 game, so it's showing as ENABLED but it's not LOADED.

Changing the Plugin Manager to show all plugins as disabled will mean that Vanilla Mode would have to disable all plugin in the PLUGINS.TXT file, meaning it will be a PERMANENT change, not a TEMPORARY one. It also makes ZERO sense, the reason you enable Vanilla Mode is because you probably have a broken plugin, and you want to be able to know which plugins are enabled even if you skipped loading them, otherwise you wouldn't even be able to test out which plugins are malfunctioning since you'd see them all as disabled.

Again, you're a retard and you shouldn't even be anywhere near ARK. Just go back to PRO or ME. Last thing I need is to keep wasting my time with the dumbest person on the planet.

Stupid people like you are the reason I had to put this in the readme:
ARK comes with no warranty whatsoever. It was designed to be noob-proof, however it is possible for the universe to create an even greater noob capable of using ARK to destroy the Earth (or his PSP). I cannot be held responsible for this.

You are that "greater noob" I was talking about. Not only are you always wrong, you actually fight against the main dev cause you think you are right, but you aren't.

Also I'm 100% sure you're a dup account of Windows10User, the dumbest user I've ever had to deal with.

RandomGuy2023, 2025-06-22:
> Abolutely NOBODY has any issues with Vanilla Mode. LITERALLY ZERO PEOPLE. The only reason YOU think it's broken is because Vanilla Mode doesn't change the text files so you keep seeing enabled [...]

If you had an IQ of 5 you'd get by now that all plugins and settings still work, even the former which work on boot.

It wasn't me who said he'd no longer reply to this thread, you fucking idiot. Then why do you keep answering me? You're not forced to do it.

If so, then why do so many people had issues with it and you keep releasing updates that fix (and introduce) them instead of only adding features, you moron?

Again, if it's me who's always wrong then why do you once in a while release updates to fix issues, jackass?

I don't know who that is, but of course, you're always right and the others are always wrong, hence the low ammount of GitHub issues created, right? Seriously, I've never met such an obtuse developer like you. Just because apparently you had an argument with another user you think others are him. I bet he is smarter than you. Anyway, will you still reply me after saying what you did?

RandomGuy2023, 2025-06-22:
> [image]

Good for you, dumbass.

JoseAaronLopezGarcia, 2025-06-22:
> [@JoseAaronLopezGarcia](https://github.com/JoseAaronLopezGarcia) can we lock this? Or ban him?

Banned.