PSP-Archive/ARK-4 #576: ISOs crashing then shutting PSP down

matheusrizzo1, 2025-05-08:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.69 r179

### Homebrew/GAME (if applicable)

UCUS-98737, ULUS-10067

### Steps to reproduce

1.  Put ISO file into /ISO folder
2. Eject PSP
3. Start game on XMB

Notes:
- Starting game through custom loader has the same error.
- I'm connecting the PSP to a Mac, don't know if this is relevant.
- sd card 32g
- Using cIPL and Full flash installation
- This is the first Ark version on this PSP

### USB Charge

On

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

On

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-05-08:
This can't be reproduced. Make sure you are safely removing the PSP.

esmith13, 2025-05-08:
I had this same issue A LOT when I first upgraded to ARK-4 from my old CFW on a PSP-3000. Drove me INSANE. Was using same ISOs as I had already. Tried changing SD Cards a few times (always SanDisk or Samsung branded cards, always a single SD Card adapter, not the 2 card version). Eventually it just stopped having the issue. I have NO IDEA what changed as it stopped crashing eventually without me even changing anything again. I had simply gotten frustrated - walked away from it a few days and then picked it up again and it "just worked".

JoseAaronLopezGarcia, 2025-05-08:
> I had this same issue A LOT when I first upgraded to ARK-4 from my old CFW on a PSP-3000. Drove me INSANE. Was using same ISOs as I had already. Tried changing SD Cards a few times (always SanDisk [...]

You probably updated it when you came back. We've been fixing many bugs, and we'll keep fixing them as they come by.

esmith13, 2025-05-08:
No I didn't. I literally fired it up to play a game and for the first time since updating to ARK-4 and the game ran without crashing the entire system. Unless the small prayer I said before booting up the PSP counts as an upgrade. 😄 

Just in case it helps to know, I'm still on the same version (r161) I was back then when all this happened to me. I was too scared to touch the CFW again after that whole fiasco. 😅

JoseAaronLopezGarcia, 2025-05-08:
> No I didn't. I literally fired it up to play a game and for the first time since updating to ARK-4 and the game ran without crashing the entire system. Unless the small prayer I said before booting up the PSP counts as an upgrade. 😄
> 
> Just in case it helps to know, I'm still on the same version (r161) I was back then when all this happened to me. I was too scared to touch the CFW again after that whole fiasco. 😅
> 
> (sorry for sorta-hijacking. I just wanted to back the OP up. I'm done now)

Only thing I can think of is hardware. There was a hardware issue that somehow solved itself (perhaps humidity, bad contacts with the memory stick, etc).

JoseAaronLopezGarcia, 2025-05-09:
No news on this? I will close it as nobody else seems to be able to reproduce, and OP doesn't seem to be replying.
OP can feel free to reopen if there's any new information.

closed 2025-05-09 by JoseAaronLopezGarcia