PSP-Archive/ARK-4 #581: stuck at "autodetecting device" then crash (PSVITA) c1-2858-3

j0s3ph19, 2025-05-10:
### PSP/Vita Model

PSP-1000 (Fat), PCH-1000/1100 (Vita Fat)

### ARK Version

r180: Nemo

### Plugins ( if applicable )

game, ms0:/seplugins/npdrm_free_mod.prx on
game, ms0:/seplugins/nploader_mod.prx on
game, ms0:/seplugins/camera_patch_lite.prx on
game, ms0:/seplugins/leda.prx on
game, ms0:/seplugins/wpa2psp.prx on 
vsh, ms0:/seplugins/npdrm_free_mod.prx on
vsh, ms0:/seplugins/nploader_mod.prx on
vsh, ms0:/seplugins/category_lite.prx on
vsh, ms0:/seplugins/cxmb.prx off
vsh, ms0:/seplugins/wpa2psp.prx on

always, bsod off

### Steps to reproduce

1 boot from xmb ARK LOADER eboot
2 the PSvita will hang at "autodetecting device" and it will crash the system to c1-2858-3 error shortly after 

### USB Charge

Off

### CPU Clock in Game

Overclock

### CPU Clock in XMB

Overclock

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

On

### Custom settings

UMD MODE: inferno 
use NoDRM engine option under adrenaline recovery menu enabled
force high memory layout option under adrenaline recovery menu disabled 
hideMAC option under adrenaline recovery menu enabled
pspemu folder stored on UMA0
6.61 Adrenaline V 7.0

note: if i try to play any video on from the XMB adrenaline completely crash/freeze (i’m not sure if it is releated to the issue or it’s just an incompatible feature)

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-05-10:
Copy the files again, make sure you safely remove the USB device before disconnecting.

closed 2025-05-10 by JoseAaronLopezGarcia

j0s3ph19, 2025-05-10:
> Copy the files again, make sure you safely remove the USB device before disconnecting.

i already done that multiple time i even checked the SHA1 and SHA256

JoseAaronLopezGarcia, 2025-05-10:
Restart the console. PS Vita can do that sometimes.

j0s3ph19, 2025-05-10:
> Restart the console. PS Vita can do that sometimes.

done, nothing changed i even checked the seplugins folder to be sure nothing was wrong, everything is correctly compiled (ex. ms0:/seplugin/plugin.prx 0)

I even copied the same files onto my friend's PS Vita, and ARK is working fine on that PS Vita (sorry if my english is not perfect is not my first language)

JoseAaronLopezGarcia, 2025-05-10:
Try using Adrenaline Bubble Maker so that ARK Loader is loaded with default settings.

j0s3ph19, 2025-05-10:
maybe it's a plugin issue. For now, I have manually disabled all plugins by setting all the values to 0 and 'off' in plugins.txt, and ARK 4 actually booted! Now I'm slowly testing the plugins. Do you know if any of my plugins could be the culprit?

j0s3ph19, 2025-05-10:
"It was Camera_Patch_Lite! I downloaded it again from [Freakler github](https://github.com/Freakler) to make sure it wasn’t just corrupted, but if this plugin is enabled ARK4 won’t boot i had to manually set it to 0 to resolve the issue

I have attached my config files.

One question: should I also enable bsod.prx on Adrenaline?

I also have a feature request, but before opening one, I want to be sure it's possible... 
so... it’s possible to play videos from XMB through Adrenaline?

Thanks for helping me troubleshoot this issue!

JoseAaronLopezGarcia, 2025-05-11:
> 
> Thanks for helping me troubleshoot this issue!

But I didn't do anything, this was all you. Congrats for figuring it out, and also thanks for the tip about camera lite plugin, I'll see if I can reproduce and fix the issue.

z-er, 2025-05-28:
Hi! I would like to  +1 this, it seems that camera patch was the reason this happens, I just happened to come across this and confirmed this was the case.

Hentilador, 2025-11-02:
Can confirm this problem happened to me too and disabling camera patch lite in autopatch 2 fixed this for me.