PSP-Archive/ARK-4 #582: .SBI support for ps1 eboots with LibCrypt

j0s3ph19, 2025-05-13:
### New Feature

i'm not entirely sure if it's possible, but there's a lot of  PS1 games that won't work at all even with PPF patches due to some LibCrypt protection, so i wonder if it is possible to add a support to .SBI subchannels file to get games like EU Formula one 99, medievil, the italian job etc etc work on PSP/Adrenaline

JoseAaronLopezGarcia, 2025-05-13:
Doesn't pop-fe already do this? Try using that.

j0s3ph19, 2025-05-13:
> Doesn't pop-fe already do this? Try using that.

yes but formula one 99 eur for example gets stuck after the first loading even with PPF patches/converted with pop-fe-ui  but it works fine if you use emulators/real hardware  i wonder if it is because is a LC4 game

closed 2025-05-14 by j0s3ph19