PSP-Archive/ARK-4 #591: Ark 4 not booting (certain?) compilation games (80010016 This game could not be started)

DJhabibi04, 2025-06-19:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

6.61 ARK-4 cIPL

### Homebrew/GAME (if applicable)

"Y's 1 and 2" "Castlevania Dracula X Chronicles"

### Plugins ( if applicable )

ms0:/seplugins/popsloader/popsloader.prx 1
ms0:/seplugins/popsloader/cdda_enabler.prx 1

### Steps to reproduce

"Y's 1 and 2" Start game from XMB, Choose either game from the in-game selection menu.

"Dracula X Chronicles": Start game from XMB. Dracula X (the main game) starts fine, but selecting either 'Symphony of The Night' or 'Rondo of Blood' from the in-game menu (after unlocking them) causes error.

This problem does not occur when I use my retail PSN digital copy of this game (ULES00841), with its corresponding license in the license folder.

Both games are ISO format.

DISABLE PSP PAUSE FEATURE IS "AUTO" (NOT AN OPTION ON YOUR DROPDOWN MENU)

### USB Charge

On

### CPU Clock in Game

Overclock

### CPU Clock in XMB

Overclock

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

On

### Prevent hibernation deletion

On

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

On

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

DJhabibi04, 2025-06-19:
Hi Jose! Thank you for the quick reply. I'm very sorry about this: I followed a Youtube guide on installing ark-4 and probably PopsLoader as well without knowing much about either of these, I'm embarrassed to say. I therefore have no particular inclination toward PopsLoader, so I will research into uninstalling it (or simply deleting it it?).

While we're here, after I have successfully uninstalled it, will I need to search for and reinstall the ARK Popcorn module? If so, would that be a simple process?

Apologies for my ignorance and thanks for your patience!

DJhabibi04, 2025-06-19:
Just to confirm Jose. I followed a guide and deleted the following file: SEPLUGINS/popsloader/popsloader.prx

The following are left intact:
SEPLUGINS/POPS.txt
SEPLUGINS/popsloader/modules (and all files and folders contained within)
SEPLUGINS/popsloader/cdda_enabler.prx
SEPLUGINS/popsloader/popscore.prx

PS1 games still work.
I also just realised that popsloader was never listed in the Plugins Manager in the recovery menu (just the example.prx).

Does this mean I've correctly removed PopsLoader?

DJhabibi04, 2025-06-19:
Hello again Jose. I'm afraid that deleting PopsLoader hasn't solved the problem and all the above issues still apply. Please let me know if I should try anything else!

DJhabibi04, 2025-06-20:
The software version is as stated (I downloaded it about 4 months ago). I have been using the in-console check for update function via wifi, but it says I'm on the latest version and there's no need to upgrade.

JoseAaronLopezGarcia, 2025-06-20:
The software version isn't stated. To view the full version you can open the VSH Menu by pressing the select button.

DJhabibi04, 2025-06-20:
Apologies! I'd put the System Software from the XMB. Updated.