PSP-Archive/ARK-4 #595: ARK_Loader Can't Detecting My Device

GrasshopperFK, 2025-06-28:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

r183

### Plugins ( if applicable )

Hello everyone, I have a PSP 1004 with OFW 6.20.
I downloaded the r183 release of ARK, but after adding the files, I can start the loader, but it crashes with the following message:

[image]

### Steps to reproduce

- Download ARK r183
- Copy the "ARK_01234" Folder into /PSP/SAVEDATA/
- Copy the "ARK_Loader" Folder into /PSP/GAME/
- Run ARK_Loader

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

krazynez, 2025-06-28:
ARK only works on 6.60 or 6.61

closed 2025-06-28 by krazynez

JoseAaronLopezGarcia, 2025-06-29:
6.20 wasn't even a good firmware. Sony broke the PS1 emulator. Also most new homebrew and plugins aim for 6.60 as it will guarantee the best compatibility across devices.