PSP-Archive/ARK-4 #597: kmod.prx crash

BradPrower, 2025-06-30:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

r184

### Steps to reproduce

While launch NandManager by Minerva

### USB Charge

On

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-06-30:
What's kmod.prx? How do you expect us to provide any help with absolutely no description of the issue.

BradPrower, 2025-06-30:
[image]

JoseAaronLopezGarcia, 2025-06-30:
What is this? This appears to be unrelated to ARK.
We have no such software in it.