PSP-Archive/ARK-4 #602: Black screen followed by an error message with a C1-2858-3 error

Databoy2023, 2025-07-04:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

4.20.69 r185

### Steps to reproduce

After installing the new standalone ARK-4 version (4.20.69 r185) I have a black screen followed by an error message with a C1-2858-3 error when running it after installing the [new](https://github.com/Kethen/NoPspEmuDrm/releases/tag/2025-07-02) NoPspEmuDrm plugin version and even the [old](https://github.com/LiEnby/NoPspEmuDrm/releases/tag/v1.05) one.

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

Databoy2023, 2025-07-04:
> Confirmed.

Could you add an option to FasterARK to only install ARK-4 so that ARK-X isn't installed, and if so, another one to only install ARK-4 and not the arkrightanalog plugin as well?

closed 2025-07-04 by JoseAaronLopezGarcia

Databoy2023, 2025-07-05:
> Confirmed.

Does the ef2uma/ef2xmc plugin need to be transferred every time a new ARK-4 version is released?

JoseAaronLopezGarcia, 2025-07-05:
No, unless the release mentions an update to it.

Databoy2023, 2025-07-05:
> No, unless the release mentions an update to it.

Now I'm having an error message with the C1-2749-2 after running the latest ARK-4 version with the latest NoPspEmuDrm plugins. It shows up right after the ChovySign logo.