PSP-Archive/ARK-4 #604: I think the recent update broke my UMD Drive

GoatWalker, 2025-07-04:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

ARK4 4.20.69 r186

### Homebrew/GAME (if applicable)

Any US or Japanese UMD

### Plugins ( if applicable )

game, ms0:/seplugins/TempAR/tempar.prx, on
game, ms0:/seplugins/pspstates/states_menu_pict_mod.prx, on
game, ms0:/seplugins/npdrm_free.prx, on
game, ms0:/seplugins/nploader.prx, on
game, ms0:/seplugins/cwcheat/cwcheat.prx, off
game, ms0:/seplugins/divapatch/divapatch.prx, off
game, ms0:/seplugins/mhp3rd_loader/mhp3patch.prx, off
game, ms0:seplugins/gta_trophies.prx, off
vsh, ms0:/seplugins/npdrm_free.prx, on
pops, ms0:/seplugins/cwcheat/cwcheatpops.prx, off

### Steps to reproduce

Just put in any UMD and they'll read, they'll boot but after the white PSP logo, it will suddenly freeze on a black screen until it turns itself off. I have tried 3 separate discs that I know work. I know it seems easy to dismiss as old hardware, the disc drive is on its way out, but this literally started as soon as I updated to the version released 3 hours ago.

### USB Charge

On

### CPU Clock in Game

Overclock

### CPU Clock in XMB

Overclock

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

On

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

GoatWalker, 2025-07-04:
I also want to note that I used the Home Button hotkey to boot into OFW and my Metal Gear Acid UMD booted as normal.

krazynez, 2025-07-04:
> I also want to note that I used the Home Button hotkey to boot into OFW and my Metal Gear Acid UMD booted as normal.

Can you hold start when booting the UMD and see if it will load.

GoatWalker, 2025-07-04:
It will show the white PSP logo when you select the UMD but it will not load the game proper. With Resistance Retribution it shows a black screen and does not load anything after. The home button does not work, the UMD stops spinning but as soon as I open the UMD door it shows the little loading icon on the bottom right corner but then freezes and shuts off completely.

JoseAaronLopezGarcia, 2025-07-04:
Completely unrelated. There are lots of people who use ARK om a daily basis to dump UMDs anad no issues has been found.

GoatWalker, 2025-07-04:
Nope, it's completely related. Why is it that when holding down the home button, my UMDs actually run, but when I let Ark4 run as is, it doesn't even run the disc? This isn't a hardware issue, it's a software issue that started when I updated the firmware today.

krazynez, 2025-07-04:
> Nope, it's completely related. Why is it that when holding down the home button, my UMDs actually run, but when I let Ark4 run as is, it doesn't even run the disc? This isn't a hardware issue, it's a software issue that started when I updated the firmware today.

Did you do what I asked you to do?

"Hold Start when booting the UMD" and keep holding it until it boots into the game

GoatWalker, 2025-07-04:
Yes I did, and it wouldn't boot the disk. When I held down the Home button, then my UMDs would run. I can even show you a video of what I did. I'll even show you a video of me holding the Start Button down too.

GoatWalker, 2025-07-04:
https://youtu.be/pLfWpSZTCC8
This first video is me repeating the steps that I said I did in the comments to get my discs to work.

https://youtu.be/vxqHFoYV8Lk
This second video shows me holding down the Start button like you said.

As for affecting some games, considering I tried Metal Slug Anthology, Metal Gear Acid and Resistance Retribution, that seems to be a very big list of games that aren't working. Also when I just acquired Metal Slug Anthology weeks before I updated my PSP today.

GoatWalker, 2025-07-04:
Also I just tried my Japanese GTA Vice City Stories that I got a month ago, tested it myself like every other game I got, and it's doing the same as the previous three games I mentioned a couple comments ago.

JoseAaronLopezGarcia, 2025-07-04:
A stupid typo got into some code and fucked shit up. It should be good now.

closed 2025-07-04 by JoseAaronLopezGarcia