PSP-Archive/ARK-4 #606: VSH MENU stops working after opening the internet browser and closing it

RegularRabbit05, 2025-07-06:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

187

### Homebrew/GAME (if applicable)

xmb

### Plugins ( if applicable )

I held start and select to disable all customization I've made

### Steps to reproduce

1. in the xmb check if the vsh menu is working by pressing select
2. open the internet browser and connect
3. go back to the xmb
4. press select and nothing happens anymore

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Custom settings

I held start and select to disable all customization I've made

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

RegularRabbit05, 2025-07-06:
Hello again, yes I'm aware that the select button is used by some applications and as such is temporarily disabled.
The problem is that for the web browser specifically it never gets re-enabled when exiting (unlike the music player which brings it back as soon as you stop the playback).
I suppose something the web browser is using remains loaded for a little longer and this https://github.com/PSP-Archive/ARK-4/blob/a73358e11af3748a532947b83a40282bd456c2e2/core/vshctrl/vshmenu.c#L152 fails somewhere.
Certain actions will bring the select menu back such as starting usb mode (which means whatever is still loaded gets unloaded).
I don't think it's the memstick getting unloaded as the force open combo always works regardless.

JoseAaronLopezGarcia, 2025-07-06:
This is not a bug, it's just how the XMB works. The weh browser is still loaded in memory, that's why it is still detected by the CFW.