PSP-Archive/ARK-4 #609: Recovery Menu will not open after a full and minimal installation

Soto-Jnthan, 2025-07-07:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.69 r188

### Steps to reproduce

1. Have access to a PSP 3000 with no CFW.
2. Download ARK-4 files for the 4.20.69 r188 release
3. Move or copy ARK_01234 folder into /PSP/SAVEDATA/ folder.
4. Move or copy ARK_Loader, ARK_cIPL and ARK_Full_Installer folders into /PSP/GAME/ folder.
5. Run the following in sequential order: ARK_Loader > ARK_cIPL > ARK_Full_Installer.
6. Remove ARK_01234, ARK_Loader, ARK_cIPL and ARK_Full_Installer from the PSP.
7. Boot the Recovery Menu by pressing select on the XMB
8. Observe how the VSH menu closes and the Recovery Menu does not boot

### USB Charge

On

### CPU Clock in Game

Overclock

### CPU Clock in XMB

Overclock

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Off

### Disable UMD Drive

On

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

krazynez, 2025-07-07:
Confirmed.

krazynez, 2025-07-08:
https://github.com/PSP-Archive/ARK-4/releases/download/v1751934272/ARK-4-v1751934272.zip

Use ARK_01234 and the Full Flash Install again ( just "install" )

Remove ARK_01234 and should be all good. Let me know or go ahead and close it when you test it yourself.