PSP-Archive/ARK-4 #616: Infrastructure mode don't work

ZjemCiKolege, 2025-07-12:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

r193

### Homebrew/GAME (if applicable)

UCES01421/ UCES-01241/ULES-01408/UCES-01312

### Plugins ( if applicable )

None

### Steps to reproduce

1.Launch game
2.Menu->online-> infrastructure mode

[image]

error in invizimals+ace combat
"8002013a"

### USB Charge

On

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Custom settings

None

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-07-12:
WPA2 is known to have stability issues.
We do not give any technical support for it.

PiporGames, 2025-07-12:
I can also confirm this is an issue on my PSP 2000 *without* WPA2.

ZjemCiKolege, 2025-07-12:
> I can also confirm this is an issue on my PSP 2000 **without** WPA2.

> He is right even if you use ARK Standalone on Vita you still can't connect to infrastructure. It seems like it only affects games that use PSN though.

For now you can install 192 until it is fixed :)

closed 2025-07-13 by ZjemCiKolege

Soto-Jnthan, 2025-07-13:
If this was confirmed to be occurring on the latest builds, I do not think it should have been closed even if there is workaround by downgrading to a previous release

ZjemCiKolege, 2025-07-13:
> If this was confirmed to be occurring on the latest builds, I do not think it should have been closed even if there is workaround by downgrading to a previous release

Already fixed in 194..