PSP-Archive/ARK-4 #617: No Icons under Extra's on PSP Street E1000

123Makefile3000, 2025-07-12:
### PSP/Vita Model

PSP-E1000 (Street)

### ARK Version

ARK 4.20.69 (any release)

### Steps to reproduce

Turn on  PSP Street E1000 console.
Go to Extra's on the XMB.

(PSP Street doesn't have any association with PSN and it's icons because it has no wifi chip, so it may be stripped from firmware, which causes this problem?)

[image]

### USB Charge

On

### CPU Clock in Game

Overclock

### CPU Clock in XMB

PowerSave

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Forced

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

On

### Turn off LEDs

Off

### Disable UMD Drive

On

### Disable Analog Stick

Off

### Enable QA Flags

On

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-07-12:
Changing to another theme (official or custom) fixes this. It's an issue with the XMB, not related to ARK.

closed 2025-07-12 by JoseAaronLopezGarcia

123Makefile3000, 2025-07-12:
Yes I see that the custom motorstorm theme does have the icons. But changing it back to classic or standard still does not have these icons. 
I understood that it would come back/ be fixed after changing it and then back to what it was before, but it does not.