PSP-Archive/ARK-4 #632: Set the default year to the same as the current build of ARK-4

HitWithSpader, 2025-07-29:
### New Feature

Basically when the PSP clock resets (like removal of battery and dead clock capacitor), set the default year to the same as the year the release of ARK-4 was built.

Sony has done this with firmware updates like 6.60 setting it to 2011 by default.
This would be a small touch that would make it easier to set the current date and time.

JoseAaronLopezGarcia, 2025-07-29:
Not a bad idea. I wonder how it would work.

JoseAaronLopezGarcia, 2025-08-05:
> [@JoseAaronLopezGarcia](https://github.com/JoseAaronLopezGarcia) This code has not been tested but it should work (because my psp does not work). Maybe by updating rebootex/main.c
> 
> New function patchSystemClock() with a call to sceRtcSetCurrentClockLocalTime() The call was inserted in _arkReboot() right after patchRebootBuffer();
> 
> [main.zip](https://github.com/user-attachments/files/21549377/main.zip)

There's gotta be an easier way, perhaps reversing an official updater to figure out how Sony does it.