PSP-Archive/ARK-4 #635: DJMax Portable 3 Not Functioning Properly

yureitzk, 2025-08-04:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

`4.20.69 r180`

### Homebrew/GAME (if applicable)

DJMax Portable 3

### Plugins ( if applicable )

Default ones only, Also tried to disable/enable ones, makes no difference in this case.

### Steps to reproduce

In your recent versions of ARK (`ark4.20.69 r180` to the latest), DJMax Portable 3 has been experiencing slower performance compared to previous versions (`ark4.20.68 r13`). This issue affects both PSP models: 2000 and 3000, as well as different memory stick formats. The game feels as though it is running slowly from a very slow memory stick.

The problem was absent in earlier versions (like `ark4.20.68 r13`) and only emerged after updating to newer ARK versions. 

**What I tried**:

- Reformatting memory stick/Using different memory stick
- Resetting `ARK` settings
- Using different PSPs (2000 and 3000)
- Trying different ROMs of the game

**Tested Firmware Versions:** 
- `6.60`
- `6.61`

This issue does not occur on alternative CFW like ProCfw.

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

Off

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-08-04:
Duplicated issue.
There are patches you can apply to the ISOs to fix them.

closed 2025-08-04 by JoseAaronLopezGarcia

yureitzk, 2025-08-04:
I couldn't find an original issue for this. Can you, please, provide me the link to where the issue was discussed previously?

yureitzk, 2025-08-21:
I examined the original issue, but it appears to be a different one, as I'm using a PSN version of the game without AP protection. [issue123](https://github.com/PSP-Archive/ARK-4/issues/124) is about AP protection and UMD emulation, but in my case the issue didn't exist in older versions of ARK, and not related to it (at least, it doesn't look like it). 

Also, [JoseAaronLopezGarcia said](https://github.com/PSP-Archive/ARK-4/issues/124#issuecomment-2771783151)

> Just download the pre-patched ISOs or the PSN version and be done with it.

So, it seems like it's a different issue, possibly related to some change in newer ARK versions.

JoseAaronLopezGarcia, 2025-10-31:
Someone else reported that the games work fine with the cfwblocker plugin.

kathyrollo, 2026-02-15:
@yureitzk Since you're the only one who seems to get it to work without the `cfwblock` plugin, the burden of proof will be on you.

If it helps your investigation, I sourced my PSN ISO from [here](https://romsfun.com/download/dj-max-portable-3-11433/4).

```
File     : DJ Max Portable 3 (USA) (PSN).iso
CRC32    : 0bf156f8
MD5      : 46f03556660a3350421e554d13423f0c
SHA1     : 5fbc6009460aa36d01f0212e62b1d6646ae9754a
```

This also verifies you're not using a pre-patched ISO just in case.