PSP-Archive/ARK-4 #637: Me & My Katamari crashes on infinite boot while first game launch loading screen

SanekOgon, 2025-08-10:
### PSP/Vita Model

PSP-E1000 (Street)

### ARK Version

v1753725903

### Homebrew/GAME (if applicable)

both european ULES-00339 and japanese ULJS-00033

### Plugins ( if applicable )

ms0:/SEPLUGINS/660_pluginfix.prx
ms0:/SEPLUGINS/passlock.prx
ms0:/SEPLUGINS/xmbih/xmbih.prx
ms0:/SEPLUGINS/category_lite/category_lite.prx
ms0:/SEPLUGINS/dayviewer.prx
ms0:/SEPLUGINS/hold.prx
ms0:/SEPLUGINS/remotejoy/RemoteJoyLite.prx
ms0:/SEPLUGINS/cxmb/cxmb.prx
ms0:/SEPLUGINS/pspdoc.prx
ms0:/SEPLUGINS/prxshot/prxshot.prx
ms0:/SEPLUGINS/mp3play/mp3play_lite.prx
ms0:/SEPLUGINS/music_autostart/music_autostart.prx

### Steps to reproduce

1. Launch game from memstick
2. It instantly crashes on first loading screen
Before i moved to ARK, game was booting on PRO-C Infinite

### USB Charge

On

### CPU Clock in Game

Overclock

### CPU Clock in XMB

PowerSave

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Forced

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-08-10:
Try disabling plugins. Also get rid of 660 plugin fix, it's known to cause issues.

JoseAaronLopezGarcia, 2025-08-10:
The ISO could be corrupted. Or the memory stick broke.
There's quite a lot of things you can try.

SanekOgon, 2025-08-10:
PPSSPP loads game image from that sd card with no problems at all thru psp's data cable

JoseAaronLopezGarcia, 2025-08-10:
Good to know, thanks.

JoseAaronLopezGarcia, 2025-09-09:
If we don't get a reply in a few days then we'll close this as unable to reproduce.