PSP-Archive/ARK-4 #646: Subtle Input Lag on All Games When Using Full Flash Installation

KarnFu, 2025-08-31:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.69 r198

### Plugins ( if applicable )

Plugins: CXMB, Dayviewer, Dayviewer Game, Custom Home, Cwcheat Per Game, Zerovsh Patcher and Leda

I have noticed a subtle but consistent input lag in all games, which I believe is caused by the "Full Flash Installation" method of the ARK-4 custom firmware. This issue manifests as unresponsive button presses or a "stiff" feeling in the controls, making games that require precise timing (like racing games or fighting games) difficult to play. The issue is not visible as a frame rate drop. I have been able to consistently resolve this by removing the full installation and using only the CIPL installation.

Steps to Reproduce:

Perform a "Full Flash Installation" of ARK-4 on the PSP.

Launch any commercial PSP game or PSOne Classic (e.g., Ridge Racer or Gran Turismo).

Play the game for a few minutes.

Notice the subtle delay in button presses and a feeling of unresponsiveness, especially during high-speed actions like drifting.

Expected Behavior:
Button presses and other inputs should be instantly responsive, with no noticeable lag, regardless of the ARK-4 installation method.

Actual Behavior:
When using the "Full Flash Installation," there is a subtle input lag that makes the controls feel unresponsive. This issue is resolved by switching to the CIPL installation.

Additional Notes:

This problem persists across multiple games and homebrew applications.

I have tried restarting the system, changing CPU clock speeds, and clearing the cache, but the only permanent solution was changing the installation method.

The issue seems to be related to the additional memory usage or background processes of the full flash installation. 

### Steps to reproduce

1. Perform Full Flash Installation of Ark-4 on the PSP
2. Launch any commercial PSP or PSOne Classic (e.g., Gran Turismo or Ridge Racer)
3. Play a game for a few minutes
4. Notice the subtle delay in button presses and a feeling of unresponsiveness, especially during high speed action games

### USB Charge

On

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

Off

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-08-31:
Well first of all, you have overclock disabled, this means that by default it'll run at 222MHz in most cases as opposed to 333MHz.

Secondly, some of the plugins you have mentioned tap into controller input to do some logic.
The PSP is a device from 2004 with 333MHz of CPU power, there's only so much it can do at once.

closed 2025-08-31 by JoseAaronLopezGarcia

KarnFu, 2025-09-02:
Thanks I took the time to go through my set up and saw what you were talking about and it all makes sense