PSP-Archive/ARK-4 #647: Console shuts off during reboot for the ark-4 loader

Loony-Chan, 2025-09-04:
When trying to boot the ark-4 loader it just crashes my PSP shuts off completely and doesn’t boot and that’s on the r198 version of ark-4 if anyone has a solution for this please let me know I appreciate it thank you

Loony-Chan, 2025-09-04:
> What version of the system are you running?

6.61 it’s on the most current software the I ran a health scan on my micro sd and it’s good I’m putting my files all in the right places as well

JoseAaronLopezGarcia, 2025-09-04:
Try reinstalling the firmware, or using one of the latest pre-releases.

Loony-Chan, 2025-09-04:
> Try reinstalling the firmware, or using one of the latest pre-releases.
That’s what I was going to try an do is install one of the pre-release and see what happens because I’ve re installed the firmware numerous times up till this point and got the same result every time

JoseAaronLopezGarcia, 2025-09-04:
What?