PSP-Archive/ARK-4 #649: what type is my cIPL?

piotr19820304-sketch, 2025-09-06:
### New Feature

as an psp 1003 user (model compatible with both types) i wonder if i have new or classic cIPL.
can you maybe add information onto system software that its new or classic in r200 release?
i use default settings. i used copilot ai and it said "well u have new cIPL, nvm its classic, or is it?"

piotr19820304-sketch, 2025-09-06:
btw i used the microsoft browser copilot, not github one.

piotr19820304-sketch, 2025-09-06:
please do not put it to "wontfixit" category thingy. i want to know when i have to buy pandora.

krazynez, 2025-09-06:
Just run the cIPL installer again you will see which

piotr19820304-sketch, 2025-09-06:
im scared to use cipl installer again but no risk no reward. btw how to exit cipl flasher without flashing anything?

krazynez, 2025-09-06:
> im scared to use cipl installer again but no risk no reward. btw how to exit cipl flasher without flashing anything?

read what it says on screen.

piotr19820304-sketch, 2025-09-06:
well thankfully i installed despartar del cementerio so i shouldnt worry about brick

piotr19820304-sketch, 2025-09-06:
how do i convert to new?

piotr19820304-sketch, 2025-09-06:
charging...

krazynez, 2025-09-06:
GitHub is meant to post issues not a message board thread. If you want to have more of a conversation you can join the PSP homebrew discord server

piotr19820304-sketch, 2025-09-06:
oh, my bad. i dont have discord :(